UZ-SLAMLab/ORB_SLAM3 #627: D435i ros stereo-inertial not IMU meas

riderpunch, 2022-07-31:
I'm using V1.0 of ORB_SLAM3 and preset d435iyaml ,already commenting out: 'sensor.set_option(RS2_OPTION_AUTO_EXPOSURE_LIMIT,50000)', not enough acceleration part and not enough acceleration and Not enough motion for initializing. Reseting... part  to make the non-ROS version of stereo-inertial work normally. 
But when I use the ros version, not IMU meas will always appear and reset map. 
I would like to know how to solve it.

CodeineX, 2023-04-23:
@riderpunch Thank you for asking this doubt, could you please tell how you approached this issue and could you fix it? I have been facing the same issue since the last week.
@jmm-montiel @jj-gomez @jdtardos @richard-elvira could you provide any inputs to this. Also I am very thankful to you for maintaining this repository.

riderpunch, 2023-04-23:
I must regret to tell you that I did not solve this problem.
I switched to opensource_tracking.launch in realsense_ros to provide similar functions, but the disadvantage is that he cannot move at high speed.
High-speed movement will cause the gyroscope to make robot_localization unable to generate the correct odom, which will cause the program positioning to fail to recover.

cyizhuo, 2023-08-28:
I think this problem is caused by "imu timestamp and img timestamp diffs too much", and can be solve by setting "global_time_enabled = true" in realsense launch file, and verified by echo imu timestamp and img timestamp