UZ-SLAMLab/ORB_SLAM3 #630: orb slam 3

morrsgarage, 2022-08-15:
Hey folks,

I'm trying to use the orb slam 3 interface to obtain positions in space, i don't have a camera physically connected, instead i pass the frames and the imu to  System::TraceRGBD and essentially obtain the information in runtime.

anyway so i am using LMWAFER docker image, the examples are working just fine but w/e i attempt to invoke this code:

   ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::IMU_RGBD, true, 0,
                           "/dpds/ORB_SLAM3/Examples/Tevel/slam-core/config/RealSense_D435i.yaml");

with 

dpds/ORB_SLAM3/build/bin/tevel_slam /dpds/ORB_SLAM3/Vocabulary/ORBvoc.txt /dpds/ORB_SLAM3/Examples/Tevel/slam-core/config/RealSense_D435i.yaml

the error:
`ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D-Inertial
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.6.0-dev) /dpds/opencv/modules/core/src/persistence.cpp:692: error: (-5:Bad argument) Input file is invalid in function 'open'

Aborted (core dumped)

please help
`

morrsgarage, 2022-08-15:
please help :CCC

morrsgarage, 2022-08-15:
i managed to pass that:

root@r131-far-7b76474d46-xbcgl:/dpds/ORB_SLAM3/build/bin# ./tevel_slam ORBvoc.txt RealSense_D435i.yaml 
2022/08/15 15:59:08.664750 [INFO] timeUtility::getCurrentMicro() USE ROS IS SET TO FALSE
2022/08/15 15:59:08.664795 [TRACE] GrpcAbstractDriver::GrpcAbstractDriver()
2022/08/15 15:59:08.664801 [TRACE] GrpcAbstractDriver::setThresholdToPrint(), sendingThreshold=80, grabbingThreshold=50
2022/08/15 15:59:08.665355 [INFO] FrontCameraDataInterface::FrontCameraDataInterface(), GOT DNS FROM ENVIRONMENT VARIABLE, [r131-front-camera-driver], IP=10.43.170.209
2022/08/15 15:59:08.665362 [INFO] FrontCameraDataInterface::FrontCameraDataInterface(), GOT PORT FROM ENVIRONMENT VARIABLE 3500
2022/08/15 15:59:08.665381 [TRACE] sharedMemoryHandler::sharedMemoryHandler(), UNIQUE IDENTIFIER- r131_frontCameraDriver, SIZE- 24, OWNER- false
2022/08/15 15:59:08.665389 [TRACE] sharedMemoryHandler::reInitSharedMemorySize(), UNIQUE IDENTIFIER- r131_frontCameraDriver, RESIZING TO=4301468
2022/08/15 15:59:08.665398 [TRACE] doubleBufferSMHandler::initAsSlave(), r131_frontCameraDriver, SUCCESSFULLY
2022/08/15 15:59:08.665942 [TRACE] sharedMemoryHandler::sharedMemoryHandler(), UNIQUE IDENTIFIER- r131_D435_GYRO, SIZE- 24, OWNER- false
2022/08/15 15:59:08.665947 [TRACE] sharedMemoryHandler::reInitSharedMemorySize(), UNIQUE IDENTIFIER- r131_D435_GYRO, RESIZING TO=238
2022/08/15 15:59:08.665952 [TRACE] doubleBufferSMHandler::initAsSlave(), r131_D435_GYRO, SUCCESSFULLY
2022/08/15 15:59:08.665957 [TRACE] sharedMemoryHandler::sharedMemoryHandler(), UNIQUE IDENTIFIER- r131_D435_ACCL, SIZE- 24, OWNER- false
2022/08/15 15:59:08.665960 [TRACE] sharedMemoryHandler::reInitSharedMemorySize(), UNIQUE IDENTIFIER- r131_D435_ACCL, RESIZING TO=238
2022/08/15 15:59:08.665965 [TRACE] doubleBufferSMHandler::initAsSlave(), r131_D435_ACCL, SUCCESSFULLY
ORBvoc.txt
RealSense_D435i.yaml

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D-Inertial
Loading settings from RealSense_D435i.yaml
Camera1.k3 optional parameter does not exist...
        -Loaded camera 1
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
        -Loaded image info
        -Loaded IMU calibration
        -Loaded RGB-D calibration
        -Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
        -Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
        -Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
        -Loaded misc parameters
----------------------------------
SLAM settings: 
        -Camera 1 parameters (Pinhole): [ 617.201 617.362 324.637 242.462 ]
        -Camera 1 distortion parameters: [  0 0 0 0 ]
        -Original image size: [ 640 , 480 ]
        -Current image size: [ 640 , 480 ]
        -Sequence FPS: 30
        -Gyro noise: 0.01
        -Accelerometer noise: 0.1
        -Gyro walk: 1e-06
        -Accelerometer walk: 0.0001
        -IMU frequency: 200
        -RGB-D depth map factor: 1000
        -Features per image: 1250
        -ORB scale factor: 1.2
        -ORB number of scales: 8
        -Initial FAST threshold: 20
        -Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Illegal instruction (core dumped)

now stuck on orb vocublary loading...

morrsgarage, 2022-08-16:
Starting program: /dpds/ORB_SLAM3/Examples/RGB-D-Inertial/rgbd_inertial_realsense_D435i /dpds/ORB_SLAM3/Vocabulary/ORBvoc.txt /dpds/ORB_SLAM3/Examples/RGB-D-Inertial/RealSense_D435i.yaml
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe7c53700 (LWP 1378)]
[New Thread 0x7fffe7452700 (LWP 1379)]
[New Thread 0x7fffe6c51700 (LWP 1380)]
[New Thread 0x7fffe6c51700 (LWP 1381)]
[Thread 0x7fffe6c51700 (LWP 1380) exited]
[Thread 0x7fffe6c51700 (LWP 1381) exited]
[New Thread 0x7fffe6c51700 (LWP 1382)]
[New Thread 0x7fffe6450700 (LWP 1383)]
[New Thread 0x7fffe5c4f700 (LWP 1384)]

[New Thread 0x7fffe544e700 (LWP 1385)]
[New Thread 0x7fffe4c4d700 (LWP 1386)]
[New Thread 0x7fffe444c700 (LWP 1387)]
[New Thread 0x7fffe3c4b700 (LWP 1388)]
[New Thread 0x7fffe344a700 (LWP 1389)]
Sensor supports the following options:

  0: Backlight Compensation is not supported
  1: Brightness is not supported
  2: Contrast is not supported
  3: Exposure
       Description   : Depth Exposure (usec)
       Current Value : 8500
  4: Gain
       Description   : UVC image gain
       Current Value : 16
  5: Gamma is not supported
  6: Hue is not supported
  7: Saturation is not supported
  8: Sharpness is not supported
  9: White Balance is not supported
  10: Enable Auto Exposure
       Description   : Enable Auto Exposure
       Current Value : 1
  11: Enable Auto White Balance is not supported
  12: Visual Preset
       Description   : Advanced-Mode Preset
       Current Value : 0
  13: Laser Power
       Description   : Manual laser power in mw. applicable only when laser power mode is set to Manual
       Current Value : 150
  14: Accuracy is not supported
  15: Motion Range is not supported
  16: Filter Option is not supported
  17: Confidence Threshold is not supported
  18: Emitter Enabled
       Description   : Emitter select, 0-disable all emitters, 1-enable laser, 2-enable laser auto (opt), 3-enable LED (opt)
       Current Value : 1
  19: Frames Queue Size
       Description   : Max number of frames you can hold at a given time. Increasing this number will reduce frame drops but increase latency, and vice versa
       Current Value : 16
  20: Total Frame Drops is not supported
  21: Auto Exposure Mode is not supported
  22: Power Line Frequency is not supported
  23: Asic Temperature is not supported
  24: Error Polling Enabled
       Description   : Enable / disable polling of camera internal errors
       Current Value : 1
  25: Projector Temperature is not supported
  26: Output Trigger Enabled
       Description   : Generate trigger from the camera to external device once per frame
       Current Value : 0
  27: Motion Module Temperature is not supported
  28: Depth Units
       Description   : Number of meters represented by a single depth unit
       Current Value : 0.001
  29: Enable Motion Correction is not supported
[... 401 lines cut ...]
	-Accelerometer noise: 0.1
	-Gyro walk: 1e-06
	-Accelerometer walk: 0.0001
	-IMU frequency: 200
	-RGB-D depth map factor: 1000
	-Features per image: 1250
	-ORB scale factor: 1.2
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...

Thread 1 "rgbd_inertial_r" received signal SIGILL, Illegal instruction.
0x00007ffff239b588 in DBoW2::FORB::fromString(cv::Mat&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /dpds/ORB_SLAM3/Thirdparty/DBoW2/lib/libDBoW2.so
(gdb) 
(gdb) backtrace
#0  0x00007ffff239b588 in DBoW2::FORB::fromString(cv::Mat&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () at /dpds/ORB_SLAM3/Thirdparty/DBoW2/lib/libDBoW2.so
#1  0x00007ffff797b109 in DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::loadFromTextFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
    at /dpds/ORB_SLAM3/lib/libORB_SLAM3.so
#2  0x00007ffff796b58d in ORB_SLAM3::System::System(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ORB_SLAM3::System::eSensor, bool, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () at /dpds/ORB_SLAM3/lib/libORB_SLAM3.so
#3  0x000055555540cc71 in main ()
(gdb) 

more info please help........... :cactus:

saadnec, 2022-10-18:
Did U solve it?

saadnec, 2022-10-18:
> i managed to pass that:
>
> root@r131-far-7b76474d46-xbcgl:/dpds/ORB_SLAM3/build/bin# ./tevel_slam ORBvoc.txt RealSense_D435i.yaml 2022/08/15 15:59:08.664750 [INFO] timeUtility::getCurrentMicro() USE [...]

How did u end up passing it?

naren2cmu, 2023-11-10:
Hi. I am also getting an illegal instruction error with ORBSLAM2. What should I do get past this error?

#0  0x0000555555605ef6 in DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::loadFromBinaryFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
#1  0x00005555555f94e5 in ORB_SLAM2::System::System(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ORB_SLAM2::System::eSensor, bool, bool, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()

amsjntz, 2025-11-05:
Have any of you ever found the problem?