UZ-SLAMLab/ORB_SLAM3 #632: CMake Error at PangolinTargets.cmake (Threads::Threads)

Manuel-Meyer, 2022-08-18:
Hi,

I am trying to generate the build files of ORB_SLAM3 on Windows with cmake and get the following error:

`CMake Error at D:/Tools/Pangolin/build/PangolinTargets.cmake:52 (set_target_properties):
  The link interface of target "pango_core" contains:

    Threads::Threads

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  D:/Tools/Pangolin/build/PangolinConfig.cmake:6 (include)
  CMakeLists.txt:48 (find_package)
`

Has anybody encountered the same error?

vyi, 2024-05-28:
@Manuel-Meyer  Have you been able to resolve this issue ?

lishiyu005, 2024-09-04:
add find_package(Threads REQUIRED) in cmake.txt will solve your problem

find_package(Threads REQUIRED)
find_package(Pangolin REQUIRED)