UZ-SLAMLab/ORB_SLAM3 #633: Sophus fails to build on Raspberry Pi

piran-prog, 2022-08-19:
Hi,
I'm trying to get this working on a raspberry pi 4 (without ROS) but it fails when trying to build sophus:
[image]

It get's this far then just sits there locking up the whole machine.

piran-prog, 2022-08-19:
It did work through in the end but failed with an error 2
[image]

SvenDC, 2022-08-19:
Might be because you run out of RAM? The system just locking up hints at it at least.
Try adding -j1 to your build command. This will limit the amount of jobs it runs concurrently and might be enough to get it to build.

piran-prog, 2022-08-20:
Brilliant thanks.
That got the sophus to build.
Unfortunately it looks like it's running out of RAM and crashing cc1plus at optimiser.c now

piran-prog, 2022-08-20:
looks like it's the same as issue  #327

piran-prog, 2022-08-20:
Fixed It.
It was a memory issue. Setting the -j1 in the build.sh helped. The default swap on the Pi is 100MB - I expanded this to 2GB and it has now compiled.

closed 2022-08-20 by piran-prog

Taoran2023, 2024-03-14:
hi, did you successfully run ORBSLAM3 on Pi4?

piran-prog, 2024-03-14:
Sort of. Yes I got it running but only through ROS, the performance was very poor and if the loop closing thread kicked off it took so much resources it effectively lost localisation.

Taoran2023, 2024-03-15:
good to hear!  I'm trying to run orbslam3 mono-inertial test base on pi4.