UZ-SLAMLab/ORB_SLAM3 #634: D455 camera in stereo mode won't initialze

antithing, 2022-08-23:
Hi, I am running on Windows, in stereo-inertial mode with a realsense d455 camera. The application launches, but i get no features in the image, and the console prints:

Initialization of Atlas from scratch
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name:
There are 1 cameras in the atlas
Camera 0 is pinhole
422 dropped frs
First KF:0; Map init KF:0
New Map created with 41 points
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
Starting the Viewer
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 1
mnInitialFrameId = 0
1 Frames set to lost
270 dropped frs

and so on. I have updated my calibration data in the yaml, what might be wrong here?

Thanks!

Czx1813260515, 2022-09-13:
hello i met the same problem,do you figure that question?

antithing, 2022-09-13:
Hi! No, I couldn't solve it. Please post here if you figure it out!

bikcrum, 2023-02-16:
@antithing Hi, I have Realsense d455 camera. Is it possible for you to share your configuration file?