UZ-SLAMLab/ORB_SLAM3 #635: How can I use OSA map file to point cloud library?

TJ-seoultech, 2022-08-24:
I'm trying to use a map file to draw a floor plan of a point cloud. And I'm planning to use it in Point Cloud Library. But the file format is an OSA format, not a normal type for 3d point cloud file like .PCD.

How can I convert an OSA map file to a usable point cloud file?

827346462, 2022-10-14:
can  you use OSA file ?

NJNischal, 2023-03-08:
Has anyone figured this or any other way to save point cloud points?

chetanpandey1266, 2023-09-01:
Has anyone figured this one or some other way ??

altaykacan, 2024-01-02:
Hey all,

I just posted my code that saves the map points in a txt file and some Python code to save the results as `.ply` files. Check it out [here](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/390#issuecomment-1874447943), I hope it helps!