UZ-SLAMLab/ORB_SLAM3 #636: ORB SLAM3 - ROS does not build!

vinsco30, 2022-08-25:
Hi guys, I am trying to use ORB SLAM3 with ROS examples. I've already installed ORB SLAM3 and all its dependecies, I can run it on datasets and with my cameras (realsense T265). Then I've tried to link it to ROS using the examples and I met some problems. The first one, following the instruction written in the readme file, I exported the folder "Example" in my bashrc file in this way: `export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:PATH/ORB_SLAM3/Examples/ROS`, adding this line below the source command of my ros workspace `source ~/catkin_ws/devel/setup.bash`. But in the last commit of the repository the folder containing the ros exmples is not under the folder "Examples" but under "Exmples_old", so I changed the export line in bashrc file in this way: `ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:PATH/ORB_SLAM3/Examples_old/ROS` and I changed, also, the file `build_ros.sh` changing the third line in `cd Examples_old/ROS/ORB_SLAM3`. Then I tried to build using this last file but it gives me this error in terminal output:
`Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:7 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:16 (cmake_policy):
  The OLD behavior for policy CMP0002 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:18 (cmake_policy):
  The OLD behavior for policy CMP0003 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:20 (cmake_policy):
  The OLD behavior for policy CMP0005 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:23 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Warning (dev) at /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:155 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    CMAKE_PROJECT_VERSION
    CMAKE_PROJECT_VERSION_MAJOR
    CMAKE_PROJECT_VERSION_MINOR
    CMAKE_PROJECT_VERSION_PATCH
Call Stack (most recent call first):
  CMakeLists.txt:4 (rosbuild_init)
This warning is for project developers.  Use -Wno-dev to suppress it.

[rosbuild] Building package ORB_SLAM3
[rosbuild] using multiarch 'x86_64-linux-gnu' for finding Boost
CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:366 (cmake_policy):
  The OLD behavior for policy CMP0037 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:4 (rosbuild_init)

-- Using CATKIN_DEVEL_PREFIX: /home/vinsco/ORB3/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/build/devel
-- Using CMAKE_PREFIX_PATH: /home/vinsco/catkin_ws/devel;/opt/ros/noetic
-- This workspace overlays: /home/vinsco/catkin_ws/devel;/opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3.8 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3.8
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() for dry packages
-- Using CATKIN_TEST_RESULTS_DIR: /home/vinsco/ORB3/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3.8 (found version "3.8.10") 
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
[rosbuild] Including /opt/ros/noetic/share/roslisp/rosbuild/roslisp.cmake
[rosbuild] Including /opt/ros/noetic/share/roscpp/rosbuild/roscpp.cmake
[rosbuild] Including /opt/ros/noetic/share/rospy/rosbuild/rospy.cmake
Build type: Release
-- Using flag -std=c++14.
CMake Error at CMakeLists.txt:37 (message):
  OpenCV > 2.4.3 not found.

-- Configuring incomplete, errors occurred!
See also "/home/vinsco/ORB3/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
See also "/home/vinsco/ORB3/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/build/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.`

I looked for other closed issues with similar problems, someone suggests to change the version od cmake (to 14), Itried this but sitl doesn't work. **Have you any suggestion to fix this problem?**

zoldaten, 2022-08-31:
cant find opencv:

> CMake Error at CMakeLists.txt:37 (message):
> OpenCV > 2.4.3 not found

just build it and try again

> .No project() command is present. 

add first string to CMakeLists.txt:
`project(ORB_SLAM3)`

closed 2022-11-07 by vinsco30

Z-star1, 2023-03-09:
sudo ln -s 软链接就可以应该