UZ-SLAMLab/ORB_SLAM3 #637: Jacobian issue in EdgePriorPoseImu?

sherman-luo, 2022-08-25:
The translation error is 
`const Eigen::Vector3d et = Rwb.transpose()*(VP->estimate().twb-twb);`

The Jacobian here is 
`_jacobianOplus[0].block<3,3>(3,3) = Rwb.transpose()*VP->estimate().Rwb;`

First of all, why define the translation error with the transformation to imu frame? What's the point of that? 
We don't do that for velocity 
`const Eigen::Vector3d ev = VV->estimate() - vwb;`

Second of all why do we have the jacobian as `Rwb.transpose() * VP->estimate().Rwb`? 
Is the jacobian not just `Rwb.transpose()`? 

I wonder why we cannot simplify it to 
`const Eigen::Vector3d et = VP->estimate().twb-twb`
jacobian as
`Identity()`

BITcats, 2022-09-05:
I have similar confusion on the Jaccbian calculation in **EdgeInertial**

The preintegration observation error about translation is

```
const Eigen::Vector3d ep = VP1->estimate().Rwb.transpose()*(VP2->estimate().twb - VP1->estimate().twb - VV1->estimate()*dt - g*dt*dt/2) - dP;
```

which can be found [here](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/src/G2oTypes.cc#L530)
The Jacobian about twb1 is

```
_jacobianOplus[0].block<3,3>(6,3) = -Eigen::Matrix3d::Identity();
```

and about twb2 is

```
_jacobianOplus[4].block<3,3>(6,3) = Rbw1*Rwb2;
```

where `Rbw1 = VP1->estimate().Rwb.transpose()` and `Rwb2 = VP2->estimate().Rwb`

I wonder why the Jacobian is not just in the following form

```
_jacobianOplus[0].block<3,3>(6,3) = -Rbw1;
_jacobianOplus[4].block<3,3>(6,3) = Rbw1
```

I try to replace jacobian with that mentioned above and test the system on MH_03 of EuRoC datasets. The frame will often be set **lost** so that the system cannot output a complete trajectory.

BITcats, 2022-09-05:
I think I can solve our confusion

**About the error definition**
It may be to avoid the influence of errors in the world coordinate system. Yaw and 3-DOF absolute position of IMU with respect to the world coordinate are unobservable in VI-SLAM system. In contrast, the 6-DOF relative pose of IMU is all observable. To this end, define the error in a body coordinate may be more reasonable.

**About about jacobian.**
In order to avoid the influence of rotation error on translation update, the translation update amount is defined in the IMU coordinate system. See [here](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/src/G2oTypes.cc#L199)
```
twb += Rwb * ut;
```
Therefore, we can caculate the jacobian about ut and get the jacobian form in the code.

closed 2022-09-05 by sherman-luo

JinXiangLai, 2023-07-02:
WOW, Thank you very very much!!!