UZ-SLAMLab/ORB_SLAM3 #638: how to build on raspberry pi - my solution

zoldaten, 2022-08-26:
as i spent a day to find out why all these "boosts,eigens, etc" do not let me to build ORB SLAM here`s my solution 
which should work on raspberry pi arm aarch64 and arm7l as i built on both systems:
```
---libraries---
sudo apt-get install libboost-all-dev libboost-dev libssl-dev libpython2.7-dev libeigen3-dev

---Pangolin---
cd ~
git clone https://github.com/stevenlovegrove/Pangolin
cd Pangolin
./scripts/install_prerequisites.sh recommended
cmake -B build -GNinja
cmake --build build
cmake --build build -t pypangolin_pip_install #no need

---opencv---
*https://qengineering.eu/install-opencv-4.5-on-raspberry-pi-4.html
wget https://github.com/Qengineering/Install-OpenCV-Raspberry-Pi-32-bits/raw/main/OpenCV-4-5-5.sh
sudo chmod 755 ./OpenCV-4-5-5.sh
./OpenCV-4-5-5.sh

---ORB_SLAM3---
git clone https://github.com/UZ-SLAMLab/ORB_SLAM3.git ORB_SLAM3
cd ORB_SLAM3
chmod +x build.sh
sed -i 's/++11/++14/g' CMakeLists.txt
./build.sh
```
if u have troubles with old opencv versions just remove em:
```
sudo apt-get purge '*opencv*'
sudo find / -name "*opencv*" -exec rm -rf {} \;
```

have a nice day!

AndrewOsswald, 2024-04-19:
nice!

abraxas4, 2024-06-10:
I'm experiencing a halt during the build process on my Raspberry Pi 4B with 4GB of RAM. Could you please let me know how much RAM your Pi has, so I can compare and see if that might be the cause of the issue?

zoldaten, 2024-06-13:
> I'm experiencing a halt during the build process on my Raspberry Pi 4B with 4GB of RAM. Could you please let me know how much RAM your Pi has, so I can compare and see if that might be the cause of the issue?

cant remember exactly, may be 8. actually u dont need that. just increase swap and use make -j1 if applicable.

abraxas4, 2024-06-26:
Thank you. Increasing the swap to 8G on the USB SSD drive (with 4G RAM) solved it

mjunsen123, 2024-07-02:
> Thank you. Increasing the swap to 8G on the USB SSD drive (with 4G RAM) solved it

May I know the performance of ORBSLAM3 in 8GB Rasp Pi?

abraxas4, 2024-07-02:
My Raspberry Pi has 4GB of RAM, and I initially tested ORBSLAM3 using only a webcam in mono mode. During the test, I observed that ORBSLAM3 utilized around 1.x GB of RAM. When using a mono webcam, there was some delay during cases of sudden lighting changes, which was also observed on a laptop running Ubuntu. Other than that, I didn't notice any significant differences in performance.

mjunsen123, 2024-07-02:
Thank you for your information. Did you have the opportunity to run stereo-inertial SLAM via EUROC / TUM on Rasp Pi? How many FPS you tested in mono btw?

dannybalz1, 2025-03-04:
Could you please let me know how the setup will change for monocular inertial mode?