UZ-SLAMLab/ORB_SLAM3 #639: How to visualize .osa map file in RViz?

ani-4nirudh, 2022-08-30:
Hi, I am new to git and don't know how to open issues here. For any inconvenience, I apologize. 

I have been able to save the binary map file. But I am in a pickle, as in how one can go about visualizing this binary file in RViz? Any thoughts? Another doubt that I have (if the binary file is visualized), should I create a publisher and a subscriber for the ROS nodes, as when I run the following commands:
```
roscore
rosrun usb_cam usb_cam_node
rosrun ORB_SLAM3 Mono ~/ORB_SLAM3/Vocabulary/ORBvoc.txt ~/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/Asus.yaml
```
and try to look at the topics with `rostopic list` it shows only the following:
```
/rosout
/rosout_agg
/usb_cam/camera_info
/usb_cam/image_raw
/usb_cam/image_raw/compressed
/usb_cam/image_raw/compressed/parameter_descriptions
/usb_cam/image_raw/compressed/parameter_updates
/usb_cam/image_raw/compressedDepth
/usb_cam/image_raw/compressedDepth/parameter_descriptions
/usb_cam/image_raw/compressedDepth/parameter_updates
/usb_cam/image_raw/theora
/usb_cam/image_raw/theora/parameter_descriptions
/usb_cam/image_raw/theora/parameter_updates

```
Basically nothing related to `PointCloud` is getting published.

Any help would be appreciated. Thanks :)

BeenToJupiter, 2024-03-16:
> Hi, I am new to git and don't know how to open issues here. For any inconvenience, I apologize.
> 
> I have been able to save the binary map file. But I am in a pickle, as in how one can go about reading this binary file in RViz? To be more clear, how can one "extract" the map points from this file and visualise the pointcloud somewhere (eg. Open3D or RViz). Any thoughts?
> 
> Any help would be appreciated. Thanks :)

Have you figured it out yet?

dev-ansh-r, 2025-04-09:
yep?? any luck on this?

dev-ansh-r, 2025-04-09:
> Hi, I am new to git and don't know how to open issues here. For any inconvenience, I apologize.
>
> I have been able to save the binary map file. But I am in a pickle, as in how one can go about [...]

Did you get something???