UZ-SLAMLab/ORB_SLAM3 #640: cant build ROS nodes on Raspbian

zoldaten, 2022-08-31:
hi !
i am on raspberry pi4 Raspbian armv7l.
I have built ORB_SLAM3 and try to build ros nodes as i have ROS noetic installed (desktop version).

1. **i have modified CMakeLists.txt as i it crashed  (/home/pi/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3)**: 
`sed -i 's/++11/++14/g' CMakeLists.txt`
also i comment out AR nodes as they wont build:
```
# Node for monocular camera (Augmented Reality Demo)
#rosbuild_add_executable(MonoAR
#src/AR/ros_mono_ar.cc
#src/AR/ViewerAR.h
#src/AR/ViewerAR.cc
#)
```
at start of a CMakeLists.txt i added:
`project(ORB_SLAM3)`

2. **modify build_ros.sh**
```
echo "Building ROS nodes"

cd Examples_old/ROS/ORB_SLAM3
mkdir build
cd build
cmake .. -DROS_BUILD_TYPE=Release -DPYTHON2_INCLUDE_DIR=/usr/bin/python3.7 -DPYTHON_EXECUTABLE=/usr/bin/python3
make -j
```

3. **add missing dependencies:**
---Sophus---
```
git clone https://github.com/strasdat/Sophus.git
cd Sophus
mkdir build && cd build && cmake .. && sudo make install
```

---fmt---
`sudo apt install libfmt-dev`

4. **before build in cmd do exports to avoid crash**:
```
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/pi/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3
source /opt/ros/noetic/setup.bash
```

i seems it starts building:
```
-- Build files have been written to: /home/pi/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/build
Scanning dependencies of target rospack_genmsg_libexe
[  0%] Built target rospack_genmsg_libexe
Scanning dependencies of target rosbuild_precompile
[  0%] Built target rosbuild_precompile
Scanning dependencies of target Mono
Scanning dependencies of target Stereo_Inertial
Scanning dependencies of target Stereo
Scanning dependencies of target Mono_Inertial
Scanning dependencies of target RGBD
[ 10%] Building CXX object CMakeFiles/Mono.dir/src/ros_mono.cc.o
[ 20%] Building CXX object CMakeFiles/Stereo_Inertial.dir/src/ros_stereo_inertial.cc.o
[ 30%] Building CXX object CMakeFiles/Stereo.dir/src/ros_stereo.cc.o
[ 40%] Building CXX object CMakeFiles/Mono_Inertial.dir/src/ros_mono_inertial.cc.o
[ 50%] Building CXX object CMakeFiles/RGBD.dir/src/ros_rgbd.cc.o
```
and crashes.

i see errors like that:
```
from /home/pi/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/src/ros_mono.cc:30:
/usr/local/include/sophus/se3.hpp:1117:7: error: redefinition of ‘class Eigen::Map<const Sophus::SE3<Scalar_>, Options>’
 class Map<Sophus::SE3<Scalar_> const, Options>
```

i think desicion is somewhere near. could you help a bit ?

swagatk, 2023-02-07:
@zoldaten : I am still facing issue while compiling the Sophus package in Step 3. I get the following build error:

```
In file included from /home/pi/Sophus/sophus/common.hpp:36,
                 from /home/pi/Sophus/sophus/types.hpp:7,
                 from /home/pi/Sophus/sophus/rotation_matrix.hpp:9,
                 from /home/pi/Sophus/sophus/so3.hpp:6,
                 from /home/pi/Sophus/sophus/rxso3.hpp:6,
                 from /home/pi/Sophus/test/core/test_rxso3.cpp:3:
/home/pi/Sophus/test/core/test_rxso3.cpp: In lambda function:
/home/pi/Sophus/sophus/common.hpp:50:30: error: declaration of ‘struct Sophus::Tests<Scalar_>::testFit()::<lambda()>::S’ shadows template parameter
 #define SOPHUS_FMT_STRING(x) FMT_STRING(x)
                              ^~~~~~~~~~
/home/pi/Sophus/sophus/common.hpp:64:15: note: in expansion of macro ‘SOPHUS_FMT_STRING’
   fmt::format(SOPHUS_FMT_STRING(description), ##__VA_ARGS__)
               ^~~~~~~~~~~~~~~~~
/home/pi/Sophus/sophus/test_macros.hpp:79:14: note: in expansion of macro ‘SOPHUS_FMT_STR’
       msg += SOPHUS_FMT_STR(descr, ##__VA_ARGS__);                             \
              ^~~~~~~~~~~~~~
/home/pi/Sophus/test/core/test_rxso3.cpp:288:9: note: in expansion of macro ‘SOPHUS_TEST’
         SOPHUS_TEST(passed, !isScaledOrthogonalAndPositive(sR_cols_swapped),
         ^~~~~~~~~~~
/home/pi/Sophus/test/core/test_rxso3.cpp:275:13: note: template parameter ‘S’ declared here
   template <class S = Scalar>
             ^~~~~
make[2]: *** [test/core/CMakeFiles/test_rxso3.dir/build.make:63: test/core/CMakeFiles/test_rxso3.dir/test_rxso3.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:206: test/core/CMakeFiles/test_rxso3.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```
That's why I am able to proceed after this step. I am testing on RPi 4 with Raspbian buster. I have installed the ROS noetic desktop version as you have suggested.  The build with '~/ORB_SLAM3/build.sh' file works just fine.
Is there anything that I might be doing wrong?  I will really appreciate some help here.

zoldaten, 2023-02-07:
sudo apt install libfmt-dev should go before build Sophus, i`ll correct.

swagatk, 2023-02-07:
This did not solve my problem either. I am still getting the same error.  In fact, I started by following the instructions given at [this](https://habr-com.translate.goog/ru/post/689168/?_x_tr_sl=ru&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=sc) page.  So, I had already installed this package. 
Is there anything else that I can try? Thanks for replying.