UZ-SLAMLab/ORB_SLAM3 #648: How to get initial gravity direction estimate during IMU initialization

manyang-yihang, 2022-09-26:
Dear @richard-elvira , @jj-gomez 

I have some confusion about [the way in the implementation to get the initial estimate of the direction of gravity](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/LocalMapping.cc#L1238)

Here are my thoughts:

Based on eq.(38) in [paper](https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=7557075) :
[image]

Multiplying both sides of the equation by the rotation matrix, we get : 
[image]

Accumulate the above formula from 0 to n - 1 and define it as -dirG :
[image]

According to this, we get the value of dirG as :
[image]

Which is a value that includes velocity and noise.

- Is it accurate to use it to estimate the direction of gravity?
- If the derivation is wrong, where can I find the correct derivation?
- If the derivation is basically correct, does this mean that the initial value estimate of the direction of gravity depends heavily on the velocity and noise during initialization? Will it lead to inaccurate estimation of gravity direction, if initialization happens at high speed?

ForLoveandPeace, 2023-03-08:
Hi, I'm also confused about it. If you have figured it out, could you teach me? Thank you!

haithink, 2024-01-17:
Here it is assumed that Vj and Vi are very close, so dirG will be close to g.