UZ-SLAMLab/ORB_SLAM3 #652: build_ros.sh still points to Examples

piran-prog, 2022-09-28:
The build_ros.sh points to the Examples folder but the files it is trying to build are now in Examples_old

SimonWXW, 2022-10-13:
You can just simply move the ROS folder from Examples_old to Examples.

piran-prog, 2022-10-13:
That is true, or update the build_ros.sh to point to the new folder, but this would have to be done by everyone who downloaded it - if it was corrected here then all new users get a working version.

SimonWXW, 2022-10-14:
Yes. 
By the way, I tried to run ORB_SLAM3 in ROS these days, but I encountered a problem--**segmentation fault** when I tried to run the monocular node.
I wonder if you have ever encountered this error and maybe have the solution?

ripdk12, 2022-10-14:
> Yes. By the way, I tried to run ORB_SLAM3 in ROS these days, but I encountered a problem--**segmentation fault** when I tried to run the monocular node. I wonder if you have ever encountered this error and maybe have the solution?

I have the same problem when i try to run the stereo_inertial node on my D435i camera, works just fine with the EuRoC dataset, and it also works when i use the EuRoC.yaml configuration (albeit very distorted obv.) which leads me to think it's got something to do with the RealSense_D435i.yaml file. I have not been able to find any information on this though

piran-prog, 2022-10-14:
I'm trying to use it with the Kinect and have found that I can only use it through ROS (at 1 frame every 4s) it gives me a segmentation fault on exit but after saving maps etc. so I haven't looked too much at it. I'll see if I can try monocular today.

SimonWXW, 2022-10-17:
> I have the same problem when i try to run the stereo_inertial node on my D435i camera, works just fine with the EuRoC dataset, and it also works when i use the EuRoC.yaml configuration (albeit very [...]

Hi, I have good news here! I just modified the OpenCV version of the whole project from 4.4 to 3.2 and then it worked!!!

fbliman, 2023-05-25:
> Hi, I have good news here! I just modified the OpenCV version of the whole project from 4.4 to 3.2 and then it worked!!!

Which ROS version are you using?
I am having the same problem

SimonWXW, 2023-05-26:
> Which ROS version are you using? I am having the same problem

ROS melodic on Ubuntu 18.04