UZ-SLAMLab/ORB_SLAM3 #660: Error while building ORB SLAM 3 for ROS

tbadave, 2022-10-17:
Hello everyone,

I have been trying to run ORB SLAM 3 for ROS (using ```build_ros.sh```) but I'm encountering the following error. I was able to build ORB SLAM 3 (using ```build.sh```) though. 

For building ORB SLAM 3 for ROS, I performed the steps as mentioned in the readme i.e.

1. Edited ```.bashrc``` file to include ```/ORB_SLAM3/Examples_old/ROS``` path to the ROS_PACKAGE_PATH environment variable.
2. ```chmod +x build_ros.sh```
3. ```./build_ros.sh```  

However, this is the error I'm seeing:

```The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
  Call Stack (most recent call first):
  CMakeLists.txt:4 (rosbuild_init)

-- Using CATKIN_DEVEL_PREFIX: /Documents/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3.8 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3.8
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() for dry packages
-- Using CATKIN_TEST_RESULTS_DIR: /Documents/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3.8 (found version "3.8.10") 
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
[rosbuild] Including /opt/ros/noetic/share/roslisp/rosbuild/roslisp.cmake
[rosbuild] Including /opt/ros/noetic/share/roscpp/rosbuild/roscpp.cmake
[rosbuild] Including /opt/ros/noetic/share/rospy/rosbuild/rospy.cmake
Build type: Release
-- Using flag -std=c++11.
CMake Error at CMakeLists.txt:37 (message):
  OpenCV > 2.4.3 not found.

-- Configuring incomplete, errors occurred! 
```
Current System:

- Ubuntu 20.04 with ROS noetic
- Opencv version - 4.6.0

Any help would be much appreciated. Thanks!

SimonWXW, 2022-10-17:
Obviously, this error occurs since OpenCV can not be detected.
The possible reasons are:
## 1.OpenCV is not installed 
To check if it is installed, run `pkg-config --modversion opencv` in the terminal to see if the version shows on the screen. If not install, you need to install OpenCV, and I suggest you can install at the version of 3.2.0. Because previously I tried to use OpenCV 4.4 to run ORB_SLAM3 in ROS and I failed.

## 2.OpenCV path is not set correctly.
Search on Google or browse the other issues under this repo to find solutions

JaimeParker, 2022-10-25:
you can read the `CMakeList.txt`, OpenCV needs a specific version of **4.4**, but if you want, you can just change it to 4 or 4.x based on the version on your pc

JaimeParker, 2022-10-26:
I met different errors
```shell
    import pyparsing as pp
  File "/opt/ros/noetic/lib/python3/dist-packages/pyparsing/__init__.py", line 100
    major: int
         ^
SyntaxError: invalid syntax
CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:77 (message):
  [rosbuild] Syntax check of ORB_SLAM2/manifest.xml failed; aborting
Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:174 (_rosbuild_check_manifest)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
```
it seems due to different python version?
I'm using ubuntu 20.04 and ros noetic

Mechazo11, 2022-10-26:
@tbadave Is your export command like the following

export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/base/Documents/ORB_SLAM3/Examples/ROS

Shivam7Sharma, 2024-05-16:
Change the CMakeList.txt in the ROS/ORB_slam folder. Add `find_package(OpenCV 4.0 REQUIRED)` and remove the 2.4.3 Opencv in the CMakeList.txt.

oskbjo, 2025-09-19:
> Change the CMakeList.txt in the ROS/ORB_slam folder. Add `find_package(OpenCV 4.0 REQUIRED)` and remove the 2.4.3 Opencv in the CMakeList.txt.

I did what @Shivam7Sharma  suggested, but I removed all of the following code block, which solved the problem for me.
```
find_package(OpenCV 3.0 QUIET)
if(NOT OpenCV_FOUND)
   find_package(OpenCV 2.4.3 QUIET)
   if(NOT OpenCV_FOUND)
      message(FATAL_ERROR "OpenCV > 2.4.3 not found.")
   endif()
endif()
```