UZ-SLAMLab/ORB_SLAM3 #661: Error while running rosrun mono

saadnec, 2022-10-18:
I have been trying to run monocular example with ros by the following command
'rosrun ORB_SLAM3 Mono Vocabulary/ORBvoc.txt Examples/Monocular/EuRoC.yaml '
but I keep getting the following error

`
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular
Segmentation fault (core dumped)
`

But when i run stereo and stereo inertial examples they are run successfully and i get the same error again while running the RGBD example

zoldaten, 2022-11-17:
have a look at publishing topics and subscribers. print em out here.

fbliman, 2023-05-25:
I am having the same problem, "Mono-inertial" "Stereo" and "Stereo-Inertial" are working, "Mono" and RGBD" fail.

I have been tracking the error but didnt find the solution yet.

This is the stack trace, in both modes I get the same error:

Thread 23 "RGBD" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff9fc63700 (LWP 320264)]
0x00007ffff6bd353e in cv::getTextSize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, double, int, int*) () from /lib/x86_64-linux-gnu/libopencv_imgproc.so.4.2
(gdb) bt
#0  0x00007ffff6bd353e in cv::getTextSize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, double, int, int*) () at /lib/x86_64-linux-gnu/libopencv_imgproc.so.4.2
#1  0x00007ffff7ade81a in ORB_SLAM3::FrameDrawer::DrawTextInfo(cv::Mat&, int, cv::Mat&) ()
    at /home/fede/Facultad/Big_files/SLAM/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#2  0x00007ffff7ae0454 in ORB_SLAM3::FrameDrawer::DrawFrame(float) ()
    at /home/fede/Facultad/Big_files/SLAM/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#3  0x00007ffff7bd00e3 in ORB_SLAM3::Viewer::Run() ()
    at /home/fede/Facultad/Big_files/SLAM/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#4  0x00007ffff762cde4 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff7dab609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#6  0x00007ffff7468133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Any clue??

dowesschule, 2023-08-05:
I have the exact same error with the ROS examples. It doesn't even start loading the vocabulary before it dumps the core. I also get no error message or stack trace, so I have no clue what could be the problem. 
The Stereo node runs just fine but Mono just intantly segfaults and dumps the core.
Have you solved the issue somehow?

also @fbliman how did you generate a stack trace?

chenzjian, 2024-02-27:
I have the exact problem with u, did u slove it? @fbliman

sirackerman, 2024-08-30:
> I have the exact problem with u, did u slove it? @fbliman

I got the same problem and solved it by first changing opencv version from 4 to 3 in the ORB_SLAM3 cmaketxt file and in the /home/user/ORB_SLAM3/Examples/ROS/ORB_SLAM3.

In those cmaketxt files, look for the line: "find_package(OpenCV 3.2 QUIET)" mine was OpenCV 4 and I changed it to 3.2 QUIET.

After that go to ORB_SLAM3 file, open the terminal and run "./build.sh", when it's completed run "./build_ros.sh. That's it.