UZ-SLAMLab/ORB_SLAM3 #664: ORB-SLAM3 suddenly works poorl

xyflzyn, 2022-10-21:
[image]
[image]
Run with your own recorded data set, start running in a straight line, then turn 180 degrees, and finally run in a straight line back to the starting point. The line was good at the beginning, but there were problems from the turn, and the result was terrible at the end. May I ask what is the cause of this?

zoldaten, 2022-11-17:
what do you use (camera,PC config)?

xyflzyn, 2022-11-17:
> what do you use (camera,PC config)?

camera：640x480
PC：11th Gen Intel(R) Core(TM) i5-11320H @ 3.20GHz   2.50 GHz

xyflzyn, 2022-11-17:
> > what do you use (camera,PC config)?
> 
> camera：640x480 PC：11th Gen Intel(R) Core(TM) i5-11320H @ 3.20GHz 2.50 GHz

This bag has a very low frame rate of about 1 frame

zoldaten, 2022-11-17:
is this mono/mono_inertial ?
1 fps is not good for algorithm. it looses the map when fast turn around.

xyflzyn, 2022-11-17:
> is this mono/mono_inertial ? 1 fps is not good for algorithm. it looses the map when fast turn around.

this is mono_inertial. The machine we used is very slow in movement and processing speed, but this problem does not occur when only using mono. When using monocular +imu, occasionally the feature points suddenly become fewer and then fail to track local map and the kind of question on the picture.

zoldaten, 2022-11-17:
camera and imu syncronised ? which fps and imu freq do you use ? what about axis of both ?

xyflzyn, 2022-11-17:
> 相机和 imu 同步？你使用哪个 fps 和 imu freq？两者的轴怎么样？

The fps of imu is 50hz. I have calibrated the camera and imu, but I am not sure whether it is accurate enough. Could you please give me some suggestions.

xyflzyn, 2022-11-17:
> > 相机和imu同步？你用哪个fps和imu freq？双方的轴怎么样？
> 
> imu 的 fps 是 50hz。我已经校准了相机和imu，但我不确定它是否足够准确。你能给我一些建议吗？

[report-imucam-image0620.pdf](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/10031098/report-imucam-image0620.pdf)

zoldaten, 2022-11-17:
i read somewhere in issues here that it is recommended to use 20 fps for camera and 200Hz for imu.
also calibration should be done with parameters above.
as far as saw in your report (used Kalibr?) you use Pinhole camera ?
could you provide your .yaml file for ORB_SLAM3 ?

xyflzyn, 2022-11-17:
> 还应使用上述参数进行校准。

I use a pinhole camera, but unfortunately the frame rate of the current camera is only about 1fps due to the limitation of hardware conditions, and I don't know whether the imu data is accurate enough. Do you have any suggestions in this regard? Thank you very much.

zoldaten, 2022-11-17:
> I use a pinhole camera, but unfortunately the frame rate of the current camera is only about 1fps due to the limitation of hardware conditions, and I don't know whether the imu data is accurate enough. Do you have any suggestions in this regard? Thank you very much.

if camera 1fps you should put this to your yaml. i see 10fps there.
but the way why do you extract 3000 features there ?! it is a large computation stress.

zoldaten, 2022-11-17:
about the imu accuracy.
if you use good enough imu and calibrated it with camera together it is enough. which imu do you use ?
if you use cheap one imu that may be a problem.
have a look my post here - https://habr.com/ru/post/689778/ 
there you can find also the link to github how to calibrate imu by itself. but in most things there is no need to do that as you use good imu.

xyflzyn, 2022-11-17:
> > 我用的是针孔摄像头，可惜目前摄像头的帧率受限于硬件条件只有1fps左右，不知道imu数据是否足够准确。在这方面你有什么建议吗？非常感谢。
> 
> 如果相机 1fps 你应该把它放到你的 yaml 中。我在那里看到 10fps。 但是为什么要在那里提取 3000 个特征？！这是一个很大的计算压力。

> > 我用摄像头，可惜可惜可惜摄像头摄像头的受限于限于条件只只只只只只只只只只只只只只只
> 
> 如果相机1fps你应该把它放到你的yaml中。我在那儿看10fps。但是为什么要在那儿取 3000个特征？！这是一个很大的计算压力。

> > 我用摄像头，可恨可恨可恨摄像头摄像头的受限仅限于物品只是只是只是只是只是只是只是只是只是只是
> 
> 如果相机1fps你应该把它放到你的yaml中。我在那儿看10fps。但是为什么要在那儿取 3000个特征？！这是一个很大的计算压力。
I am now running ORB-SLAM3 on an MR133 processor, due to heavy computation, low frame rate and instability, sometimes the interval between two frames is about 500ms, sometimes it is 1-2s, what should be set in yaml for this situation? The reason for the extraction of 3000 feature points is that 1000 would lead to too few feature points and fail to track local map.

zoldaten, 2022-11-17:
try to decrease feature points to 1500 max. increase fps to 30 or higher in camera node and in yaml ORB_SLAM. try to increase piramid level in yaml from 8 to 12.

xyflzyn, 2022-11-17:
> 尝试将特征点减少到最大 1500。在相机节点和 yaml ORB_SLAM 中将 fps 增加到 30 或更高。尝试将 yaml 中的金字塔级别从 8 增加到 12。
Thank you for your answer. I'll try it. About the calibration, I feel that there are still some problems in the previous calibration results, especially the three quantities representing the position relationship in the external parameter matrix, and a lot of curves produced are not quite the same as the example of the webpage you sent, but this has been repeated for many times or failed to solve. Do you think it is possible that the data of imu we use cannot reach the accuracy requirements of ORBSLAM3?

zoldaten, 2022-11-17:
hard to say. it is a comlicated issue. try options we disscuss above first.
and try to listen to imu topic with echo command. i remember that some of imu`s failed at low publish rate printing errors of FIFO or somewhat.

zoldaten, 2022-11-17:
see `rostopic echo /imu`
or you do not use if ?
ros helps you to understand what is going on with imu and so on.
but of picture above it looks like imu drops or not started.

xyflzyn, 2022-11-17:
> 看到`rostopic echo /imu` 还是不使用 ROS？ ros帮助你了解imu是怎么回事等等。 但是上面的图片看起来像 imu 下降或没有启动。

> rostopic echo /imu
Thank you. I will try as soon as possible. Thanks again.

xyflzyn, 2022-11-26:
@zoldaten Excuse me, Sir. This is the document I have calibrated with kalibr recently. Could you please help me check if there is any problem.
[report-imucam-.1125_cam_imu.pdf](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/10095425/report-imucam-.1125_cam_imu.pdf)

zoldaten, 2022-11-28:
@xyflzyn try to compare your result with others on Kalibr github. If calibration is well done you will see difference in inference.

xyflzyn, 2022-11-29:
@zoldaten 
[image]
Dear Sir, thank you very much for your reply. This is the calibration result of comparison, and there is an obvious difference. Do you know the reason?

Aishkrish18, 2023-06-20:
Can someone tell me if its possible to check in realtime the mapping and localisation from ORBSLAM3? or should we record the data and perform localisation offline all the time?

Aishkrish18, 2023-07-04:
I have a quick question. I would like to know how to check the realtime pose of the camera upon ORBSLAM localisation. From ROS implementation, it seems that it does not publish any pose and I would like to get some insights on how to do that. 

Thanks in advance