UZ-SLAMLab/ORB_SLAM3 #667: Description for camera transformation in Realsense_T265 YAMLs

luisvillaran, 2022-10-26:
It seems the comment for this transformation is wrong

`# Transformation matrix from right camera to left camera`
`Stereo.T_c1_c2: !!opencv-matrix`

c1 is the left camera and c2 is the right according to the Calibration_Tuturial.pdf.  This description threw me off for a while until I noticed the translation for all these files had a positive x translation, so (I think) the description should be swapped.

luisvillaran, 2022-10-26:
Actually now I"m confused... the T_c1_c2 should be the transformation taking a point from camera 2 to camera 1 (from my reading of the wiki)  but all the example YAML files have a positive x transformation. How can that be true if camera 1 is defined to be the left camera?

JoshuaU01, 2023-11-28:
I think the mistake is in the comment for the transformation between the camera and the IMU:
`# Transformation from body-frame (imu) to left camera
IMU.T_b_c1: !!opencv-matrix`

As you mentioned, this would mean a transformation from c1 to b (according to their paper _Calibration Tutorial for ORB-SLAM3 v1.0_). It seems that this is a common naming convention (i.e. _kalibr_ works the same way).

When I perform multi-cam calibration on the Realsense T265 using kalibr, I get an x value of approximately **-0.06** in the **T_cn_cnm1 matrix**. As this matrix transforms points from the last camera to the current camera (you can read this in the kalibr wiki: _https://github.com/ethz-asl/kalibr/wiki/yaml-formats_; here: left cam to right cam), it seems that the coordinate system is just "flipped" (meaning going to the right equals a negative x value).
With respective to that it makes sense to me that you have to invert this matrix now, if you want to use your calibration in orb_slam3 (because orb_slam3 goes the other way by performing a transformation from the right cam to the left cam).

So the comment for the transformation between cameras should be correct, but not so the comment for the transformation between camera and IMU.