UZ-SLAMLab/ORB_SLAM3 #668: build failure

where2go947, 2022-10-27:
Hi, I'm trying to install ORB_SLAM3 on ubuntu 20.04.5 LTS.

I have followed [the github](https://github.com/egdw/ORB_SLAM3_Ubuntu20.04) to
- update Cmakelists.txt from `-std=c++11` to `-std=c++14`
- replace all `#include <Eigen/(any packages)>` to `#include <eigen3/Eigen/(any packages)>`

The `DBoW2` and `g2o` have been built successfully, but when I configured and built ORB_SLAM3 as following:
```
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make -j
```
Error occurs: (There are many errors here, omitted for brevity, full in next)
```
Scanning dependencies of target g2o
[  4%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/types/types_seven_dof_expmap.cpp.o
[  4%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/types/types_sba.cpp.o
[  4%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/hyper_graph_action.cpp.o
[  4%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/types/types_six_dof_expmap.cpp.o
[  6%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/marginal_covariance_cholesky.cpp.o
[  7%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/hyper_graph.cpp.o
[ 10%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/batch_stats.cpp.o
[ 10%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/cache.cpp.o
[ 10%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/parameter.cpp.o
[ 12%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/optimizable_graph.cpp.o
[ 13%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/matrix_structure.cpp.o
[ 14%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/solver.cpp.o
Scanning dependencies of target ORB_SLAM3
[ 15%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_factory.cpp.o
[ 17%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/estimate_propagator.cpp.o
[ 18%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/factory.cpp.o
[ 19%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/sparse_optimizer.cpp.o
[ 20%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/System.cc.o
[ 21%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/Tracking.cc.o
[ 23%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o
[ 24%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/hyper_dijkstra.cpp.o
[ 25%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o
[ 26%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/parameter_container.cpp.o
[ 28%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/optimization_algorithm.cpp.o
[ 29%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_with_hessian.cpp.o
[ 30%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/ORBextractor.cc.o
[ 31%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/ORBmatcher.cc.o
[ 32%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_levenberg.cpp.o
[ 34%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/FrameDrawer.cc.o
[ 35%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_gauss_newton.cpp.o
[ 36%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/jacobian_workspace.cpp.o
[ 37%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/Converter.cc.o
[ 39%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/MapPoint.cc.o
[ 40%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/KeyFrame.cc.o
[ 41%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/Atlas.cc.o
[ 42%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/robust_kernel.cpp.o
[ 43%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/robust_kernel_factory.cpp.o
[ 45%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/core/robust_kernel_impl.cpp.o
[ 46%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/MapDrawer.cc.o
[ 47%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/Map.cc.o
[ 48%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o
[ 50%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/Frame.cc.o
[ 51%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/stuff/timeutil.cpp.o
[ 52%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/stuff/property.cpp.o
[ 53%] Building C object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/stuff/os_specific.c.o
[ 54%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/stuff/string_tools.cpp.o
[ 56%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/KeyFrameDatabase.cc.o
[ 57%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/Sim3Solver.cc.o
[ 58%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/Initializer.cc.o
[ 59%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/Viewer.cc.o
[ 62%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/G2oTypes.cc.o
[ 62%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/ImuTypes.cc.o
[ 63%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/CameraModels/Pinhole.cpp.o
[ 64%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/OptimizableTypes.cpp.o
[ 65%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/CameraModels/KannalaBrandt8.cpp.o
[ 67%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/MLPnPsolver.cpp.o
[ 68%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/TwoViewReconstruction.cc.o
......
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:232: CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:390: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

The report of `cmake .. -DCMAKE_BUILD_TYPE=Release` said it seems to be related to `realsense2`
but In [this issue](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/458#issuecomment-1007077761), it should be addressed by C++14
```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type: Release
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using flag -std=c++14.
-- Found OpenCV: /usr (found suitable version "4.2.0", minimum required is "4.0") 
OPENCV VERSION:
4.2.0
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
CMake Warning at CMakeLists.txt:54 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

-- BUILD TYPE:Release
-- Compiling on Unix
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lufan/Projects/xsj/learn/ORB_SLAM3_Ubuntu20.04-master/build
```

Could you give me some help? 
Thanks!!!

where2go947, 2022-10-27:
[install_log.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/9877716/install_log.txt)

where2go947, 2022-10-27:
After following the installation of [librealsense2](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#:~:text=Building%20librealsense2%20SDK)

The report of `cmake .. -DCMAKE_BUILD_TYPE=Release` seems right:
```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type: Release
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using flag -std=c++14.
-- Found OpenCV: /usr (found suitable version "4.2.0", minimum required is "4.0") 
OPENCV VERSION:
4.2.0
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- BUILD TYPE:Release
-- Compiling on Unix
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lufan/Projects/xsj/learn/ORB_SLAM3_Ubuntu20.04-master/build
```
But the error remains.
Please help : )

uedummy, 2022-11-04:
Hi, did you resolve you issue? Would be glad about any help :)

where2go947, 2022-11-17:
Sorry, fail still.

zoldaten, 2022-11-17:
have a look here:
https://answers.ros.org/question/363889/intel-realsens-on-ubuntu-2004-ros-noetic-installation-desription/
it marked as deprecated but i have built successfully realsense2, ORB_SLAM3 on odroid xu4 ubuntu 20.04.

kenhuang1964, 2023-11-15:
Hey @where2go947 , did you end up resolving the issue?