UZ-SLAMLab/ORB_SLAM3 #681: coordinate system of ORB_SLAM3 ?

zoldaten, 2022-11-19:
hi !
following the Paper i saw this picture:
[image]
camera axis:
Y camera looks - down;
X camera - right;
Z camera - forward.

IMU axis:
Y camera looks - right;
X camera - up;
Z camera - forward.

so the questions are:
1.should the whole system of camera-imu be calibrated with Kalibr as above picture says or
the axis of both should be in the same directions ?
2.which distance from camera to imu should be (to mount on robot or somewhere) and can it be changed ?

WFram, 2022-11-20:
If you use Kalibr for Camera-IMU extrinsic calibration, in order to run ORB-SLAM3 in Stereo-Inertial mode you need to set two matrices:
1) `IMU.T_b_c1`: this is the transformation from left camera frame to IMU frame (left camera pose w.r.t. IMU frame). In report from Kalibr it corresponds to `T_ic:  (cam0 to imu0)`.
2) `Stereo.T_c1_c2`: this is the transformation from right camera to left camera (right camera pose w.r.t. left camera frame). To get this transformation, the `Baseline (cam0 to cam1)` from Kalibr should be inverted. Here assumed the left camera corresponds to `cam0`.

zoldaten, 2022-11-22:
@WFram thanks.
so there is no matter how i will place IMU (in which direction) before calibration  matrices will do the job  after calibration ?

WFram, 2022-11-24:
No matter. I guess you can run it even without a prior knowledge about IMU axes orientation itself. Because the only thing you need is to know a relative transformation from camera to IMU, which can be provided by Kalibr (as an example).

zoldaten, 2022-11-24:
got it. thanks!

closed 2022-11-24 by zoldaten