UZ-SLAMLab/ORB_SLAM3 #688: build_ros.sh

hyeons75, 2022-11-28:
Ubuntu 20.04 noetic
Dual booting on my laptop
I had added  export ROS_PACKAGE_PATH
==========================================
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:7 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:16 (cmake_policy):
  The OLD behavior for policy CMP0002 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:18 (cmake_policy):
  The OLD behavior for policy CMP0003 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:20 (cmake_policy):
  The OLD behavior for policy CMP0005 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:23 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

[rosbuild] Building package ORB_SLAM3
[rosbuild] Error from directory check: /opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/hsc/Dev/ORB_SLAM3/Examples/ROS/ORB_SLAM3
1
Traceback (most recent call last):
  File "/opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:99 (message):
  [rosbuild] rospack found package "ORB_SLAM3" at "", but the current
  directory is "/home/hsc/Dev/ORB_SLAM3/Examples/ROS/ORB_SLAM3".  You should
  double-check your ROS_PACKAGE_PATH to ensure that packages are found in the
  correct precedence order.
Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/hsc/Dev/ORB_SLAM3/Examples/ROS/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found. stop.

Jhonan01, 2022-12-02:
> # Ubuntu 20.04 noetic
> Dual booting on my laptop
> I had added export ROS_PACKAGE_PATH
> CMake Warning (dev) in CMakeLists.txt: No project() command is present. The top-level CMakeLists.txt file must contain a literal, direct call to the project() command. Add a line of code such as
> 
> ```
> project(ProjectName)
> ```
> 
> near the top of the file, but after cmake_minimum_required().
> 
> CMake is pretending there is a "project(Project)" command on the first line. This warning is for project developers. Use -Wno-dev to suppress it.
> 
> CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:7 (cmake_policy): The OLD behavior for policy CMP0011 will be removed from a future version of CMake.
> 
> The cmake-policies(7) manual explains that the OLD behaviors of all policies are deprecated and that a policy should be set to OLD only under specific short-term circumstances. Projects should be ported to the NEW behavior and not rely on setting a policy to OLD. Call Stack (most recent call first): CMakeLists.txt:2 (include)
> 
> CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:16 (cmake_policy): The OLD behavior for policy CMP0002 will be removed from a future version of CMake.
> 
> The cmake-policies(7) manual explains that the OLD behaviors of all policies are deprecated and that a policy should be set to OLD only under specific short-term circumstances. Projects should be ported to the NEW behavior and not rely on setting a policy to OLD. Call Stack (most recent call first): CMakeLists.txt:2 (include)
> 
> CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:18 (cmake_policy): The OLD behavior for policy CMP0003 will be removed from a future version of CMake.
> 
> The cmake-policies(7) manual explains that the OLD behaviors of all policies are deprecated and that a policy should be set to OLD only under specific short-term circumstances. Projects should be ported to the NEW behavior and not rely on setting a policy to OLD. Call Stack (most recent call first): CMakeLists.txt:2 (include)
> 
> CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:20 (cmake_policy): The OLD behavior for policy CMP0005 will be removed from a future version of CMake.
> 
> The cmake-policies(7) manual explains that the OLD behaviors of all policies are deprecated and that a policy should be set to OLD only under specific short-term circumstances. Projects should be ported to the NEW behavior and not rely on setting a policy to OLD. Call Stack (most recent call first): CMakeLists.txt:2 (include)
> 
> CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:23 (cmake_policy): The OLD behavior for policy CMP0011 will be removed from a future version of CMake.
> 
> The cmake-policies(7) manual explains that the OLD behaviors of all policies are deprecated and that a policy should be set to OLD only under specific short-term circumstances. Projects should be ported to the NEW behavior and not rely on setting a policy to OLD. Call Stack (most recent call first): CMakeLists.txt:2 (include)
> 
> [rosbuild] Building package ORB_SLAM3 [rosbuild] Error from directory check: /opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py /home/hsc/Dev/ORB_SLAM3/Examples/ROS/ORB_SLAM3 1 Traceback (most recent call last): File "/opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in raise Exception Exception CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:99 (message): [rosbuild] rospack found package "ORB_SLAM3" at "", but the current directory is "/home/hsc/Dev/ORB_SLAM3/Examples/ROS/ORB_SLAM3". You should double-check your ROS_PACKAGE_PATH to ensure that packages are found in the correct precedence order. Call Stack (most recent call first): /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location) CMakeLists.txt:4 (rosbuild_init)
> 
> -- Configuring incomplete, errors occurred! See also "/home/hsc/Dev/ORB_SLAM3/Examples/ROS/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log". make: *** No targets specified and no makefile found. stop.

mamdouhmostafa, 2023-02-05:
Hello, have you reached a solution for this error?

Z-star1, 2023-03-09:
i have the same problem

yaoxingzhi, 2023-03-27:
Hello,have you slove the problem?i have the same problem

fbliman, 2023-04-21:
same problem, any solution?

knightwzh, 2023-05-26:
Using `source /opt/ros/melodic/setup.bash` may help, and you may change "melodic" according to your path.

Aishkrish18, 2023-07-04:
> Using `source /opt/ros/melodic/setup.bash` may help, and you may change "melodic" according to your path.

I am currently using Melodic docker and ros implementation works perfect on that. Earlier I had spend quite sometime on trying to make it work on Noetic, I had the same errors as above but I wasn't sure how to solve it. I am assuming the support is fine only for ROS Melodic and its broken for noetic

balkNfalkN, 2024-01-22:
Before the build command, try setting `export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/home/hsc/Dev/ORB_SLAM3/Examples/ROS/ORB_SLAM3`.

This is of course not a solution but a workaround that might get you through the build.