UZ-SLAMLab/ORB_SLAM3 #699: ORB-SLAM3 not working in ROS

giricoooo, 2022-12-14:
When I try to execute build_ros.sh script, the get the following error message:

-----------------------------------
```
mkdir: cannot create directory ‘build’: File exists
[rosbuild] Building package ORB_SLAM3
[rosbuild] Error from directory check: /opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/user/ORB_SLAM3/Examples/ROS/ORB_SLAM3
1
Traceback (most recent call last):
  File "/opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:99 (message):
  [rosbuild] rospack found package "ORB_SLAM3" at "", but the current
  directory is "/home/user/ORB_SLAM3/Examples/ROS/ORB_SLAM3".  You should
  double-check your ROS_PACKAGE_PATH to ensure that packages are found in the
  correct precedence order.
Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:5 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/user/ORB_SLAM3/Examples/ROS/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found. Stop.
```
-----------------------------------

To fix the problem, I opened the .bashrc file and added an `export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:PATH/ORB_SLAM3/Examples/ROS `to the last line, but got an error message like the one above.
My .bashrc file looks like this:
```
source /opt/ros/noetic/setup.bash
source ~/catkin_ws/devel/setup.bash
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:PATH/ORB_SLAM3/Examples/ROS

```

I hope someone can offer a solution.

zoldaten, 2022-12-23:
have a look my post here - https://habr.com/ru/post/689168/

Mechazo11, 2023-01-21:
https://github.com/thien94/orb_slam3_ros use this one,I have been using this since last November to build our SLAM pipeline in a real-time robot