UZ-SLAMLab/ORB_SLAM3 #700: Empty IMU measurements vector!!!

yriuss, 2022-12-14:
I get this message when running a custom dataset, but when I use "rostopic echo", everything seems fine with the "imu" topic (and I didn't forget to remap). What this error even mean?

gtech005, 2022-12-27:
Hello, I'm trying to port the ros wrapper to ros2. And I'm facing the same issue when I launch the orbslam3 using the ros wrapper with the EuroC dataset. However, it works fine without the ros wrapper. 
@yriuss , do you have any lead on this issue? Thanks!

HJMGARMIN, 2023-03-06:
Hello, have you solved the problem? I met the same issuse when I run the mode of RGB-D-Inertial.

d4n13lx13, 2024-01-16:
Have the same issue as well, when using a custom ros2 wrapper with custom data, euroc data works fine.

yriuss, 2024-01-16:
Until today I haven't solved this issue. Actually I completely changed what I was doing before. When I return to use ORB-SLAM and found any solution to this issue I will let you know.

Mingtao312, 2025-07-13:
Hello everyone, I solved this problem, You can use a higher frame rate IMU (400/1000Hz), the issue will not exist anymore.