UZ-SLAMLab/ORB_SLAM3 #719: Segmentation fault (core dumped) when running ORB_SLAM3 on RGB-D TUM sequence freiburg2_large_no_loop

zarathacia, 2023-03-03:
The command;
./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt ./Examples/RGB-D/TUM1.yaml ~/Datasets/rgbd_data ./Examples/RGB-D/associations/fr2.txt

The run:
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D
Loading settings from ./Examples/RGB-D/TUM1.yaml
    -Loaded camera 1
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
    -Loaded image info
    -Loaded RGB-D calibration
    -Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
    -Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
    -Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
    -Loaded misc parameters
----------------------------------
SLAM settings:
    -Camera 1 parameters (Pinhole): [ 517.306 516.469 318.643 255.314 ]
    -Camera 1 distortion parameters: [  0.262383 -0.953104 -0.005358 0.002628 1.16331 ]
    -Original image size: [ 640 , 480 ]
    -Current image size: [ 640 , 480 ]
    -Sequence FPS: 30
    -RGB-D depth map factor: 5000
    -Features per image: 1000
    -ORB scale factor: 1.2
    -ORB number of scales: 8
    -Initial FAST threshold: 20
    -Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name:
There are 1 cameras in the atlas
Camera 0 is pinhole

-------
Start processing sequence ...
Images in the sequence: 3357

First KF:0; Map init KF:0
New Map created with 950 points
Gtk-Message: 16:53:07.493: Failed to load module "canberra-gtk-module"
Starting the Viewer
Segmentation fault (core dumped)

I need to access the generated map but the maps are not saved and i get this error (core dumped). can you plz help me fix this? i am new to this

SJTUlibochen, 2023-03-04:
have a look at this issue [https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/452](url)

zarathacia, 2023-03-05:
sorry the link doesn't work

zarathacia, 2023-03-06:
When running the debug i got this message:

Thread 1 "rgbd_tum" received signal SIGSEGV, Segmentation fault.
0x00007ffff7e94748 in ORB_SLAM3::Frame::ComputeStereoFromRGBD(cv::Mat const&) ()
   from /home/fossa/Dev/ORB_SLAM3/lib/libORB_SLAM3.so

zarathacia, 2023-03-06:
Solved the problem was with the association file it had an extra line with only the image reference

closed 2023-03-06 by zarathacia

Flee-postgra, 2023-08-17:
> 

pls,i have confornt this problem,，may i ask you how to solved this problems?

zarathacia, 2023-08-19:
my problem originated from a faulty image depth association file. In this case check the file if there is an extra line at the end where an RGB image has no associated depth image.