UZ-SLAMLab/ORB_SLAM3 #723: Merging maps and bad point cloud?

danielj-genesis, 2023-03-11:
I'm trying to do dense reconstruction from SLAM output and I'm wondering why the point cloud is so much better with SFM.

I know it's not exactly the same but I exported the camera poses and point cloud from [ORB-SLAM2](https://github.com/raulmur/ORB_SLAM2) using Monocular setting on the  [freiburg1_room](https://cvg.cit.tum.de/data/datasets/rgbd-dataset/download#freiburg1_room) sequence.

I'm wondering if my export is wrong because the sparse point cloud looks pretty different from SFM.
[image]
vs 
[image]

I'm wondering if I exported the data incorrectly or if the video sequence is too short for ORB-SLAM to create a proper sparse cloud?

Here are the densified versions:
[image]
vs 
[image]

SFM is producing a much much better Point cloud. Is this a normal result? And if so is there a way I can extract better points? 
Alongside that ORB-SLAM3 maintains multiple maps in an atlas but I want to maintain an updated dense point cloud so losing a map ruins the data? Is there any way to not create multiple maps or to merge them more quickly?

DimitrisKatsatos, 2023-07-27:
Hi @danielj-genesis, 

I encountered the exact same problem. I extracted the sparse point cloud along with the 3D points to feed them to OpenMVS for surface reconstruction. However, the SfM pipelines seem to generate a much denser point cloud than ORB-SLAM3. I'm not sure if this is solely related to ORB features instead of SIFT/SURF. Did you find the problem?

chetanpandey1266, 2023-09-03:
What is SFM? Can you please share the links to its github