UZ-SLAMLab/ORB_SLAM3 #727: BUG report: In func "Track()"

iDonghq, 2023-03-17:
code line 1874 about func 'PreintegrateIMU()', code snippet in  'PreintegrateIMU()' from line 1678 to 1681:
'
.....
if(n==0){
        cout << "Empty IMU measurements vector!!!\n";
        return;
    }
.....
'
The func  'Track()' will continue from func 'PreintegrateIMU()' with "mCurrentFrame.mpImuPreintegratedFrame = nullptr", it leads to crashed when the function 'PredictStateIMU()'  run in the rest part of "Track()".

crashed happens in  'PredictStateIMU()' with 'mCurrentFrame.mpImuPreintegratedFrame = nullptr ',  line 1770
'
.........
const float t12 = mCurrentFrame.mpImuPreintegratedFrame->dT;
.........
'

ccl-1, 2023-03-20:
I have the same problem, have you solved this?
I use  my own dataset.

iDonghq, 2023-03-21:
> I have the same problem, have you solved this? I use my own dataset.
Change the  RETURN type for 'PreintegrateIMU()' from “void” to "bool", if 'false', return from func ''Track()'; Actually maybe some error happens in your sensor for the special case~