UZ-SLAMLab/ORB_SLAM3 #734: Relocalization mode is not working with IMU+CAMERA ( INTERIAL MONO )

S0UL4, 2023-03-24:
Hi,

Actually, orbslam3 keeps creating a new map over the loaded one and never re-localize itself on top of the previous loaded MAP... 
I'am using the MONO INTERIAL..

Is it due to the IMU or something else?

Thanks ion advance

LeFaucon-dev, 2023-03-30:
Hi,

Well... Bad news for you, I don't think Relocalization in a loaded map with ORB-SLAM3 using IMU is currently implemented. Lately, I tried to do the same thing as you and the conclusion of my countless hours of researches on this problem came to this last conclusion.

Hope it helps you move on...

S0UL4, 2023-03-30:
@jdtardos can you please confirm ? 

Thanks in advance..

LoveStorys, 2023-05-10:
@S0UL4 Hey, brother, I think I have encountered the same confusion as you. I tried to load the offline map from the outside, and then completed the pure positioning function according to the offline map, but I failed after many attempts. You have Any good ideas or suggestions? Thanks

drlssl, 2023-07-27:
> @jdtardos can you please confirm ?
> 
> Thanks in advance..

I met the same problem now, could u give me any advice ?  (●'◡'●)