UZ-SLAMLab/ORB_SLAM3 #737: Applying the changes from [this fork](https://github.com/dz306271098/ORB_SLAM3) seems to solve my problem.

LightInDust, 2023-03-27:
Applying the changes from [this fork](https://github.com/dz306271098/ORB_SLAM3) seems to solve my problem. 
Specifically, replace the `Map::PreSave` function and `MapPoint::PreSave` function with the following:

```c++
void Map::PreSave(std::set<GeometricCamera*> &spCams)
{
    int nMPWithoutObs = 0;

    std::set<MapPoint*> tmp_mspMapPoints1;
    tmp_mspMapPoints1.insert(mspMapPoints.begin(), mspMapPoints.end());

    for(MapPoint* pMPi : tmp_mspMapPoints1)
    {
        if(!pMPi || pMPi->isBad())
            continue;

        if(pMPi->GetObservations().size() == 0)
        {
            nMPWithoutObs++;
        }
        map<KeyFrame*, std::tuple<int,int>> mpObs = pMPi->GetObservations();
        for(map<KeyFrame*, std::tuple<int,int>>::iterator it= mpObs.begin(), end=mpObs.end(); it!=end; ++it)
        {
            if(it->first->GetMap() != this || it->first->isBad())
            {
                pMPi->EraseObservation(it->first);
            }

        }
    }

    // Saves the id of KF origins
    mvBackupKeyFrameOriginsId.clear();
    mvBackupKeyFrameOriginsId.reserve(mvpKeyFrameOrigins.size());
    for(int i = 0, numEl = mvpKeyFrameOrigins.size(); i < numEl; ++i)
    {
        mvBackupKeyFrameOriginsId.push_back(mvpKeyFrameOrigins[i]->mnId);
    }

    // Backup of MapPoints
    mvpBackupMapPoints.clear();

    std::set<MapPoint*> tmp_mspMapPoints2;
    tmp_mspMapPoints2.insert(mspMapPoints.begin(), mspMapPoints.end());

    for(MapPoint* pMPi : tmp_mspMapPoints2)
    {
        if(!pMPi || pMPi->isBad())
            continue;

        mvpBackupMapPoints.push_back(pMPi);
        pMPi->PreSave(mspKeyFrames,mspMapPoints);
    }

    // Backup of KeyFrames
    mvpBackupKeyFrames.clear();
    for(KeyFrame* pKFi : mspKeyFrames)
    {
        if(!pKFi || pKFi->isBad())
            continue;

        mvpBackupKeyFrames.push_back(pKFi);
        pKFi->PreSave(mspKeyFrames,mspMapPoints, spCams);
    }

    mnBackupKFinitialID = -1;
    if(mpKFinitial)
    {
        mnBackupKFinitialID = mpKFinitial->mnId;
    }

    mnBackupKFlowerID = -1;
    if(mpKFlowerID)
    {
        mnBackupKFlowerID = mpKFlowerID->mnId;
    }

}
```
and
```c++
void MapPoint::PreSave(set<KeyFrame*>& spKF,set<MapPoint*>& spMP)
{
    mBackupReplacedId = -1;
    if(mpReplaced && spMP.find(mpReplaced) != spMP.end())
        mBackupReplacedId = mpReplaced->mnId;

    mBackupObservationsId1.clear();
    mBackupObservationsId2.clear();
    // Save the id and position in each KF who view it
    std::map<KeyFrame*,std::tuple<int,int> > tmp_mObservations;
    tmp_mObservations.insert(mObservations.begin(), mObservations.end());

    for(std::map<KeyFrame*,std::tuple<int,int> >::const_iterator it = tmp_mObservations.begin(), end = tmp_mObservations.end(); it != end; ++it)
    {
        KeyFrame* pKFi = it->first;
        if(spKF.find(pKFi) != spKF.end())
        {
            mBackupObservationsId1[it->first->mnId] = get<0>(it->second);
            mBackupObservationsId2[it->first->mnId] = get<1>(it->second);
        }
        else
        {
            EraseObservation(pKFi);
        }
    }

    // Save the id of the reference KF
    if(spKF.find(mpRefKF) != spKF.end())
    {
        mBackupRefKFId = mpRefKF->mnId;
    }
}
```

_Originally posted by @heiwang1997 in https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/443#issuecomment-1154814254_

LightInDust, 2023-03-27:
I've replace my replace the Map::PreSave function and MapPoint::PreSave function as #443 showed.And I could save map without seg fault anymore.
But when I load the map in order to use the localization mode,there is no map point in the viewer.
First I guess it just some bugs in viewer.cc and has no influence on the localization mode.However,the localization will show no match and loop in reseting map.

LightInDust, 2023-03-27:
[image]
[image]

dionkrith, 2023-09-06:
I have the same problem. Is there any solution?

Sadegh-Kalami, 2023-09-07:
I did what the repo said to build orbslam3 but I have no visualization like u have shared with us. would you pls tell me how did you do this?

LightInDust, 2023-09-07:
Actually I did not solve the problem. As I said before,, there are some bugs in viewer.cc and localization mode. I tried in the code in the project: 
https://github.com/lturing/ORB_SLAM3_ROS,and it uses the RViz to achieve the visualization of ORB3.When I loaded the map ,RViz would show the point.
However,the loop in the localization mode still have some bugs.It seems that when you enter the localization mode,you will loop in somewhere I can't figure out.
Because I have left the work for almost  4 months, these are what I can do.Hope that someone will solve the problem for us in the future!