UZ-SLAMLab/ORB_SLAM3 #741: ORB SLAM3 Using Mobile Device

hanseongbugi, 2023-04-12:
I will use SAMSUNG Note 5 to measure the data and then run ORB SLAM3.
This work is being done for academic research.
I performed the calibration of IMU and Camera through the kalib that the author told me in the paper appendix.
This task allowed me to get the Tbc matrix. And I put Kalib's output in the yaml file.
Also, I measured the ORB SLAM3 data through a mobile device and processed the data according to the set format.
However, the ORB SLAM3 system fell into a loop in the initialization and Tracking.
Please tell me how to solve the problem I wrote above.

hanseongbugi, 2023-04-18:
The IMU of the mobile device exists inside the device.
Also, the location of the IMU is unknown and cannot be rid.
Therefore, could you give me a specific answer that means to rid it

hanseongbugi, 2023-04-18:
I thought it was more than a mobile device, so I used Dataset provided by Kalibr.
In other words, after extracting the image and IMU sensor values from the IMU-CAM.bag file of Kalib, IMU-Camera Calibration was performed. 
As a result, we could have a Tbc matrix and camera parameters.
I organized this value in the .yaml file and created a frame time stamp in the file for ORB SLAM3.
However, despite using the correct Tbc matrix values and IMU sensor values, this showed different results from the MH01 dataset.
Can you tell me what else I need to run ORB SLAM3 besides the Frame image and IMU sensor values, Tbc matrix, and frame time stamp?

hanseongbugi, 2023-04-26:
This issue could be resolved by creating .csv and .yaml files under the dataset directory. I'll close the issue.

deepfine-kskim, 2023-04-26:
@hanseongbugi 
Can I share the csv and yaml files?