UZ-SLAMLab/ORB_SLAM3 #742: Errors during installation

EduardoQD, 2023-04-21:
I'm trying to install the last version of orbslam3 in ubuntu 20.04.
I've installed the last version from openv concretly opencv 4.7.0.
I've installed pangolin as their repository guides.
Before that I' ve installed other depencies with these commands:

```
sudo add-apt-repository "deb http://security.ubuntu.com/ubuntu xenial-security main"
sudo apt update

sudo apt-get install build-essential
sudo apt-get install cmake git libgtk2.0-dev pkg-config libavcodec-dev libavformat-dev libswscale-dev

sudo apt-get install python-dev python-numpy libtbb2 libtbb-dev libjpeg-dev libpng-dev libtiff-dev libdc1394-22-dev libjasper-dev

sudo apt-get install libglew-dev libboost-all-dev libssl-dev

sudo apt install libeigen3-dev
```

And when I launch de compiler in ORBSLAM3 the terminal sends me this error:
```
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:89: CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:63: CMakeFiles/ORB_SLAM3.dir/src/System.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:102: CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:76: CMakeFiles/ORB_SLAM3.dir/src/Tracking.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:711: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```
Does anyone have an idea to solve it?

Vlados09, 2023-04-22:
I had this when trying to install with Pangolin v0.8, try downgrading to Pangolin v0.6:

```
cd opt/ && \
 git clone https://github.com/stevenlovegrove/Pangolin.git Pangolin && \
 cd Pangolin/ && \
 git checkout 0.6 && \
 mkdir build && \
 cd build/ && \
 cmake -D CMAKE_BUILD_TYPE=RELEASE -D CPP11_NO_BOOST=1 .. && \
 make -j$(nproc) && \
 make install
 ```

EduardoQD, 2023-04-25:
It solved the errors but I have some problems with using D435 becuase the CMAKE file didn't find the realsense2.

NontawatWuttikam, 2023-06-17:
> 

This worked after hours of fixing. How did you get a clue to downgrade Pangolin?

EduardoQD, 2023-06-19:
I have solved it installing the correct commit of pangolin for this version

EduardoQD, 2023-06-19:
> Do you have librealsense2 installed? If so, how did you install it? As binaries with apt or from source?

Now it is working.