UZ-SLAMLab/ORB_SLAM3 #747: Is there a way of notifying that a loop closure has been performed?

wjdghksdl26, 2023-04-24:
First of all, thank you all for bringing orb-slam3; it works wonderfully for my environment.

With that said i would like to ask if there is a way of notifying the completion of a loop closure.

I've tried keeping track of the variable mLastLoopKFid of the LoopClosing class, but it doesn't seem to change in spite of obvious loop closures.

Any way of notifying a new loop closure would suffice.

Any help would be appreciated!

Mechazo11, 2023-09-05:
@wjdghksdl26 I was able to inform that the calculations for a loop closure was completed by printing out a message from the `void LoopClosing::CorrectLoop()` method in the LoopClosing.cc

wjdghksdl26, 2023-09-19:
thank you @Mechazo11, that worked for me.

closed 2023-09-19 by wjdghksdl26