UZ-SLAMLab/ORB_SLAM3 #748: D435 error launching

EduardoQD, 2023-04-25:
Hello, 
 I have a D435 without IMU and I'm trying to launch the Monocular example with this command,
`./Examples/Monocular/mono_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/Monocular/RealSense_D435i.yaml`

I'm getting this error in the terminal:
terminate called after throwing an instance of 'rs2::invalid_value_error'
what(): object doesn't support option #85 
Aborted ('core' generated)

I don't know if the issue its the files becuase I'm using a different model of D435i .

knightwzh, 2023-05-26:
Hi, I'm using D435i and facing the same problem. But at my case, it's running well in ROS

awaistahir29, 2023-07-17:
> Is the problem resolved? 
I am trying to use RealSense D435 which is without IMU information. Can you tell how does it work with this ORB-SLAM3?

tyz1030, 2023-07-27:
same issue here

IAMHAADICOOL, 2023-11-08:
Yeah, can you please provide a solution?

IAMHAADICOOL, 2023-12-15:
I lowered the library version to 2.50 and it worked, although stereo - inertial ones are not working.
It's throwing error- not enough acceleration
Any idea how to fix that?

Cheng-Nan-Lee, 2023-12-15:
Sorry to say but I have not encountered such a problem myself.