UZ-SLAMLab/ORB_SLAM3 #749: Empty IMU measurements vector, Not preintegrated measurement

hanseongbugi, 2023-04-29:
``
$ ./Monocular-Inertial/mono_inertial_euroc ../Vocabulary/ORBvoc.txt ./Monocular-Inertial/Note5.yaml ../DataSets/Note_01/ ../DataSets/Note_01/frame_timestamps.txt
num_seq = 1
Loading images for sequence 0...LOADED!
Loading IMU for sequence 0...LOADED!

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name:

Camera Parameters:
- Camera: Pinhole
- Image scale: 1
- fx: 480.5330810546875
- fy: 479.59561157226562
- cx: 310.04620361328125
- cy: 193.75247192382812
- k1: 0.18193915486335754
- k2: -0.22170035541057587
- p1: 0.0076599419116973877
- p2: -0.0042910254560410976
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters:
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2000000476837158
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Left camera to Imu Transform (Tbc):
[0.074703753, -0.99630672, -0.042335421, -9.0560003e-05;
 -0.99641669, -0.076265901, 0.036568899, 0.00021031;
 -0.039662588, 0.03945189, -0.99843401, -6.061e-05;
 0, 0, 0, 1]

IMU frequency: 100 Hz
IMU gyro noise: 0.0030000000260770321 rad/s/sqrt(Hz)
IMU gyro walk: 1.9392999092815444e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 0.00016999999934341758 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.0030000000260770321 m/s^3/sqrt(Hz)
There are 1 cameras in the atlas
Camera 0 is pinhole
error: XDG_RUNTIME_DIR not set in the environment.
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
First KF:0; Map init KF:0
New Map created with 445 points
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Starting the Viewer
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Empty IMU measurements vector!!!
Not preintegrated measurement
Segmentation fault
``

JulienRineau, 2024-12-04:
Hi @hanseongbugi, did you found any solution for this ? I'm having the same issue

hanseongbugi, 2024-12-04:
Hi @JulienRineau, Samsung Note 5's camera shoots images in a rolling shutter manner.
As a result, the exact Tic matrix cannot be obtained during the calibration between the IMU and the camera.
It can be solved by using a global shutter-type camera.

closed 2024-12-04 by hanseongbugi