UZ-SLAMLab/ORB_SLAM3 #754: Stuck on trying to initialize - no features detected on image viewer

nikkusan, 2023-05-21:
I'm running ORB SLAM 3 on a turtlebot waffle pi which is simulated in gazebo
ORB SLAM 3 's image viewer is able to  receive the images published from the turtlebot but no features are detected.
Although I've moved the turtlebot around it says TRYING TO INITALIZE  on the image viewer and in the map viewer, it's blank with 1 green rectangle. It doesn't move when I move the turtlebot.

I've also gotten the error howeven even when installing the gtk module with apt-get I still get this error and I have read somewhere that I am able to ignore the GTK error
(ORB-SLAM3: Current Frame:2124): dbind-WARNING **: 06:21:47.750: Couldn't connect to accessibility bus: Failed to connect to socket /run/user/1000/at-spi/bus_0: No such file or directory
Gtk-Message: 06:21:47.792: Failed to load module "canberra-gtk-module"
Gtk-Message: 06:21:47.796: Failed to load module "canberra-gtk-module"

fbliman, 2023-08-28:
I am having exact same problem, any solution?

nicko369, 2023-08-29:
> I am having exact same problem, any solution?

What are you simulating this on?
Is it on motion?

As I understand it, the GTK errors can be ignored. 
The reason why there's no features is because ORBSLAM3 is trying to initialise. 

So perform some movements on the Z axis (looking left and right) to begin initialisation and some features should appear.