UZ-SLAMLab/ORB_SLAM3 #757: Realsense D435i stuck on "Waiting for images"

hseysen, 2023-05-24:
I'm attempting to run the ROS version of ORB_SLAM3, and I want to test it in real-time using my Realsense D435i camera. I've not changed the Monocular-Inertial .yaml file of the camera calibration, since I assume it would be more or less similar (Although I'm not entirely sure if my camera is producing frames at 640x480 or not).

I've run the ros camera in one terminal using the following command:
```bash
roslaunch realsense2_camera rs_camera.launch unite_imu_method:=copy
```

The last option allowed me to combine accelerometer and gyroscope into one topic called `/camera/imu`, which I made sure to subscribe to in the `ros_mono_inertial.cc` file.

```c++
  // Maximum delay, 5 seconds
  ros::Subscriber sub_imu = n.subscribe("/camera/imu", 100, &ImuGrabber::GrabImu, &imugb); 
  ros::Subscriber sub_img0 = n.subscribe("/camera/color/image_raw", 15, &ImageGrabber::GrabImage,&igb);
```

Next, I ran this command on another terminal to start ORB_SLAM3 ROS:
```bash
rosrun ORB_SLAM3 Mono_Inertial Vocabulary/ORBvoc.txt Examples/Monocular-Inertial/RealSense_D435i.yaml false
```

And here I am stuck at "Waiting for images", and I just can't seem to find the source of the problem. I've opened another terminal to run `rqt_graph`, which shows me that `/Mono_Inertial` has successfully subscribed to the relevant topics:
[image]

How could I approach the solution on this? Perhaps the resolution is really not 640x480 in my case (according to [this answer](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/501#issuecomment-1057448236) that was the root cause of their problem). I'm still new at ROS, so I would appreciate any help

horry0314, 2023-07-04:
Excuse me. Have you figured out this problem? I have the same problem.

wengchenyang1, 2023-08-01:
I think the issue might be related to either `StereoInitialization()` or `MonocularInitialization()` (in `Tracking.cc`).

If the failure of the initialization is vision related, possible solutions might be moving the camera to scenes with more feature points when the program starts. Also, because of the "Waiting for images" issue, we can not check the quality of the captured image (e.g., maybe exposure is too low), so we may use opencv to show the image before sending it to SLAM. Another quick check is to reduce the values of 
```
ORBextractor.iniThFAST
ORBextractor.minThFAST
```
in the config file and see if it helps.

hseysen, 2023-08-15:
After having wasted so much time on trying different things, I decided to recalibrate the RealSense D435i camera, and to my surprise, the `.bag` file that I recorded for Kalibr did not pick up the `/camera/imu` topic. I assume this was a hardware failure, and will move on to fusing my own camera and imu.