UZ-SLAMLab/ORB_SLAM3 #767: How to publish the pose in real time？

lalawinter, 2023-06-14:
I want to publish the obtained pose in real time，So I changed the code in ros_mono_inertial.cc:

    cv::Mat Rwc = Tcw.rowRange(0, 3).colRange(0, 3).t();
    cv::Mat twc = -Rwc * Tcw.rowRange(0, 3).col(3);
    vector<float> q = ORB_SLAM3::Converter::toQuaternion(Rwc);

    poseMSG.pose.position.x = -twc.at<float>(0);
    poseMSG.pose.position.y = -twc.at<float>(2);
    poseMSG.pose.position.z = twc.at<float>(1);
    poseMSG.pose.orientation.x = q[0];
    poseMSG.pose.orientation.y = q[1];
    poseMSG.pose.orientation.z = q[2];
    poseMSG.pose.orientation.w = q[3];
    poseMSG.header.frame_id = "VSLAM";
    poseMSG.header.stamp = ros::Time::now();
Tcw is obtained by mpSLAM->TrackMonocular(im,tIm,vImuMeas)；

I tested it on the EuRoc MH03 sequence, but the difference between the pose and the true value is relatively large；
So how to publish the pose in real time correctly?

Shang-cl, 2023-07-04:
@lalawinter 
I also encountered the same problem. I was trying to output the pose information in the case of binoculars, using mpSLAM->TrackStereo(imLeft, imRight, tImLeft, vImuMeas); the return matrix of this function, but I found The output positioning information is very different from the actual positioning information. It seems that the positioning information obtained by using this function needs to be multiplied by 3 to be close to the correct positioning information. Do you have any better suggestions or solutions?

Mechazo11, 2023-09-05:
@lalawinter Monocular SLAM does not have scale information. So the pose data you are obtaining is based on an estimated scale not the true scale. Without additional data, as far as I know, monocular SLAM cannot figure out correct scale properly.