UZ-SLAMLab/ORB_SLAM3 #778: How to correctly output positioning information?

Shang-cl, 2023-07-04:
I am using a stereo camera + IMU for the positioning of the map construction. I am now using mpSLAM->TrackStereo(imLeft, imRight, tImLeft, vImuMeas); to obtain the positioning information, but I found that the output positioning information is not correct, for example, in the fixed The z-axis position of the actual coordinate information of the current position in the scene is 10, but the value of the positioning information obtained by using this function on the Z-axis is 3, and the difference between the two is about 3 times. Why does this happen? question?

closed 2023-07-10 by Shang-cl

ShengpingY, 2024-08-21:
Hi, @Shang-cl ,I am currently also facing this problem, trying yo find out the proper variant for postion of camera, did you find a proper solution for this?

Shang-cl, 2024-08-22:
> 你好，@Shang-cl,我目前也面临这个问题，试图找出适合相机位置的正确变体，你找到合适的解决方案了吗？

I think this (project)[https://github.com/thien94/orb_slam3_ros] is more better.