UZ-SLAMLab/ORB_SLAM3 #779: ROS stereo-inertial ORB_SLAM3 producing "segmation fault (core dumped)" crash randomly during operation

s4561341, 2023-07-05:
Hi

First off thanks for all the hard work releasing ORB_SLAM3!

After numerous failed attempts of building all dependencies and the algorithm itself (with no prior experience of programming), I finally got the algorithm running on my ubuntu 18.04 machine. Sadly after recording my own dataset with an T265 and using the stereo-inertial ROS node to stream data to, the ORB_SLAM3 algorithm randomly produces the "Segmentation fault (core dumped) error" with no extra information regarding the error. Weirdly this can occur on the start, near the end or not at all during the SLAM process which makes it very hard to determine the exact reason for this error. Has somebody experienced this before and knows a way to fix this?

I've tried adjusting the settingsfile with no to little result next to looking at solutions specified by other users but no issue seems to match my error. The only thing in my mind that can cause the error is the rosbag playrate as this is set to 0.5 because the algorithm on my setup can't handle the 30fps datastream (the "Camera.fps" and "IMU.Frequency" parameters are adjusted accordingly) but I'm not an expert.

Your help is much appreciated!

Error:
_First KF:1; Map init KF:0
New Map created with 30 points
start VIBA 1
end VIBA 1
start VIBA 2
end VIBA 2
Segmentation fault (core dumped)._

akkoyunlu, 2024-06-10:
how did you solve?

isaiahoh, 2024-07-23:
Hey, I also came across this problem, using gdb it seems that the error comes when there there is less than 3 VisEdges for a keyframe. 

`ORB_SLAM3/src/Optimizer.cc:2837: static void ORB_SLAM3::Optimizer::LocalInertialBA(ORB_SLAM3::KeyFrame*, bool*, ORB_SLAM3::Map*, int&, int&, int&, int&, bool, bool): Assertion `mit->second>=3' failed.`

which leads to the thread aborting:
`Thread 21 "stereo_inertial" received signal SIGABRT, Aborted.`

I am unsure of what parameters to change .yaml to ensure that this doesn't happen, any help would be greatly appreciated!

khshmt, 2024-07-23:
Try build it with release build_type not debug

isaiahoh, 2024-07-23:
I built before release, got the error, then switched to debug, to see where the error was coming from. I think issue maybe the data I am using is for a vehicle so not enough movement for IMU to initialize well. Any thoughts?

khshmt, 2024-07-24:
I do not think not enough movement is the problem, because if it is so the orb SLAM algorithm will reset map but not crash.

I do not know how you get this error in release build (ORB_SLAM3/src/Optimizer.cc:2837: static void ORB_SLAM3::Optimizer::LocalInertialBA(ORB_SLAM3::KeyFrame*, bool*, ORB_SLAM3::Map*, int&, int&, int&, int&, bool, bool): Assertion ) is it an assertion which will not appear in release build it is only debug or you could try "#define NDEBUG" before "#include<assert.h>" in "optimzer.cc"

isaiahoh, 2024-07-24:
@khshmt Oh sorry, I should've been more clear, when I state the error I meant the original posted one:
> start VIBA 1
> end VIBA 1
> start VIBA 2
> end VIBA 2
> Segmentation fault (core dumped).
and I wanted to back trace to see where the error is, so went into debug mode. I still get the segmentation fault in release mode, I run the `build.sh` file, as I am not using ROS. 

Do you suggestions for methods I can use to track down the issue without going into debug mode, I am still learning how to efficiently code and debug in C++. Thanks in advance!