UZ-SLAMLab/ORB_SLAM3 #781: I successfully compiled orb-slam3 and was able to run the mh01 dataset. But when I run my webcam I get a "segmentation fault, core dumped" error. I deleted -march=native in cmakelist and recompiled and still encountered the same error. How to solve it?

wotingfengchuiyu, 2023-07-13:
orbslam3@orbslam3-virtual-machine:~/ORB_SLAM3$ rosrun ORB_SLAM3 Mono /home/orbslam3/ORB_SLAM3/Vocabulary/ORBvoc.txt /home/orbslam3/ORB_SLAM3/Examples/ROS/ORB_SLAM3/Asus.yaml

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- Image scale: 1
- fx: 535.4
- fy: 539.2
- cx: 320.1
- cy: 247.6
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
There are 1 cameras in the atlas
Camera 0 is pinhole
Starting the Viewer
段错误 (核心已转储)

drlssl, 2023-07-19:
same problem, have u solved this ?

JoannaZhj, 2023-08-16:
same problem QQ, trying to solve it