UZ-SLAMLab/ORB_SLAM3 #793: How to publish pose topic on ROS2 with monocular -inertial node?

astronaut71, 2023-08-08:
Hi 

I checked the post for the mono-inertial node of ros1 with [https://github.com/pinkfloydfan/ORB_SLAM3/blob/83d41a1e924c6b9e45ce0e2aca7398107c3bb111/Examples/ROS/ORB_SLAM3/src/ros_mono_inertial.cc#L206] . Now try to adapt for ros2 humble and also to publish the pose. Here is my function where it should happen

```
...

void ImageGrabber::SyncWithImu()
{

  while (rclcpp::ok()) 
  {
    cv::Mat im;
    double tIm = 0;
    if (!img0Buf.empty()&&!mpImuGb->imuBuf.empty())
    {
      tIm = img0Buf.front()->header.stamp.sec + img0Buf.front()->header.stamp.nanosec * 1e-9; 
      if (tIm > mpImuGb->imuBuf.back()->header.stamp.sec + mpImuGb->imuBuf.back()->header.stamp.nanosec * 1e-9) 

          continue;
      {
        this->mBufMutex.lock();
        im = GetImage(img0Buf.front());
        img0Buf.pop();
        this->mBufMutex.unlock();
      }

      vector<ORB_SLAM3::IMU::Point> vImuMeas;
      mpImuGb->mBufMutex.lock();
      if(!mpImuGb->imuBuf.empty())
      {
        // Load imu measurements from buffer
        vImuMeas.clear();
        while (!mpImuGb->imuBuf.empty() && mpImuGb->imuBuf.front()->header.stamp.sec + mpImuGb->imuBuf.front()->header.stamp.nanosec * 1e-9 <= tIm) 

        {
          double t = mpImuGb->imuBuf.front()->header.stamp.sec + mpImuGb->imuBuf.front()->header.stamp.nanosec * 1e-9;          
          cv::Point3f acc(mpImuGb->imuBuf.front()->linear_acceleration.x, mpImuGb->imuBuf.front()->linear_acceleration.y, mpImuGb->imuBuf.front()->linear_acceleration.z);
          cv::Point3f gyr(mpImuGb->imuBuf.front()->angular_velocity.x, mpImuGb->imuBuf.front()->angular_velocity.y, mpImuGb->imuBuf.front()->angular_velocity.z);
          vImuMeas.push_back(ORB_SLAM3::IMU::Point(acc,gyr,t));
          mpImuGb->imuBuf.pop();
        }
      }
      mpImuGb->mBufMutex.unlock();
      if(mbClahe)
        mClahe->apply(im,im);

      cv::Mat T_, R_, t_ ;
      //Sophus::SE3f T_;

      //stores return variable of TrackMonocular
      //cv::Mat TrackMonocular(const cv::Mat &im, const double &timestamp, const vector<IMU::Point>& vImuMeas = vector<IMU::Point>(), string filename="");
      T_= mpSLAM->TrackMonocular(im,tIm,vImuMeas);

        if (!(T_.empty())) 
        {

          RCLCPP_INFO(logger_, "Enering publisher execution block");

          cv::Size s = T_.size();
          if ((s.height >= 3) && (s.width >= 3))
           {
              R_ = T_.rowRange(0,3).colRange(0,3).t();
              t_ = -R_*T_.rowRange(0,3).col(3);
              std::vector<float> q = ORB_SLAM3::Converter::toQuaternion(R_);
              float scale_factor=1.0;
              tf2::Transform tf_transform;

              tf_transform.setOrigin(tf2::Vector3(t_.at<float>(0, 0) * scale_factor, t_.at<float>(0, 1) * scale_factor, t_.at<float>(0, 2) * scale_factor));
              tf_transform.setRotation(tf2::Quaternion(q[0], q[1], q[2], q[3]));

              //RCLCPP_INFO(this->get_logger(), "Mono_Inertial node started");
              RCLCPP_INFO(logger_, "Transformation matrix R_: %s", cv::format("%f %f %f\n%f %f %f\n%f %f %f",
                R_.at<float>(0, 0), R_.at<float>(0, 1), R_.at<float>(0, 2),
                R_.at<float>(1, 0), R_.at<float>(1, 1), R_.at<float>(1, 2),
                R_.at<float>(2, 0), R_.at<float>(2, 1), R_.at<float>(2, 2)).c_str());            

          }
        }

    }

    //RCLCPP_INFO(logger_, "After publisher execution block");
    std::chrono::milliseconds tSleep(1);
    std::this_thread::sleep_for(tSleep);
  }
}
...
```

The code has one problem as need to convert Cv::Mat type to Sophus::SE3. So I would like to publish the pose of the ORB_SLAM3 using camera and IMU. The problem is in this line

`T_= mpSLAM->TrackMonocular(im,tIm,vImuMeas);`

And I got this error:

    `error: no match for ‘operator=’ (operand types are ‘cv::Mat’ and ‘Sophus::SE3f’ {aka ‘Sophus::SE3’}) 257 | T_= mpSLAM->TrackMonocular(im,tIm,vImuMeas);`

So how is the way to implement CvMatToSE3 function , mean this conversion? Or do I need to implement the CvMatToSE3 function in the Converter class (inside Converter.h in ~ORB_SLAM3/include/Converter.h or a separate header/cpp file)?

y2d2, 2023-09-23:
Try this: 

```
Sophus::SE3f  T_ = SLAM_->TrackMonocular(im, tIm, vImuMeas);
cv::Mat T_cv = cv::Mat(4, 4, CV_32F, T_.matrix().data());
```

it worked for me.

Nikolaiarn, 2024-01-23:
Hi

@astronaut71 I am also working on implementing the mono-inertial node for ros2 using this wrapper https://github.com/zang09/ORB_SLAM3_ROS2/tree/main/src
But it is also missing mono-inertial node? Did you get it to work? 
if so, do you mins sharing the code?

Nikolaiarn, 2024-01-28:
@astronaut71 @y2d2 @Unknown9190 Did any of you get the mono_inertial node to work? I have been stuck on this for weeks. If you can share the code it would be much appreciated

astronaut71, 2024-03-06:
I tried many version of this node nut none working. maybe you can fix it 

```
#include <iostream>
#include <queue>
#include <thread>
#include <mutex>

#include "rclcpp/rclcpp.hpp"
#include <cv_bridge/cv_bridge.h>
#include "sensor_msgs/msg/imu.hpp"
#include "sensor_msgs/msg/image.hpp"
#include <opencv2/core/core.hpp>
#include <opencv2/imgproc/imgproc.hpp>
#include <Eigen/Dense>
#include <opencv2/core/eigen.hpp>
#include "utility.hpp"

#include <System.h>
#include "ImuTypes.h"
#include" Converter.h"

#include "tf2_ros/transform_broadcaster.h"
#include "tf2/LinearMath/Transform.h"
#include "tf2_geometry_msgs/tf2_geometry_msgs.h"
#include <geometry_msgs/msg/pose_stamped.hpp>

using namespace std;

class ImuGrabber
{
public:
    ImuGrabber(const rclcpp::Logger& logger)
        : logger_(logger) {
        // Constructor implementation
    }

    void GrabImu(const sensor_msgs::msg::Imu::SharedPtr imu_msg);    
    queue<sensor_msgs::msg::Imu::SharedPtr> imuBuf;
    std::mutex mBufMutex;

private:
    rclcpp::Logger logger_;
};

class ImageGrabber
{
public:
    ImageGrabber(ORB_SLAM3::System* pSLAM, ImuGrabber *pImuGb, const bool bClahe, rclcpp::Logger logger)
        : mpSLAM(pSLAM), mpImuGb(pImuGb), mbClahe(bClahe), logger_(logger) {}

    void GrabImage(const sensor_msgs::msg::Image::SharedPtr img_msg);
    cv::Mat GetImage(const sensor_msgs::msg::Image::SharedPtr img_msg);
    void SyncWithImu();
    void SetPub(rclcpp::Publisher<geometry_msgs::msg::PoseStamped>::SharedPtr pub);

    queue<sensor_msgs::msg::Image::SharedPtr> img0Buf;
    std::mutex mBufMutex;

    ORB_SLAM3::System* mpSLAM;
    ImuGrabber *mpImuGb;
    
    bool mbClahe;
    cv::Ptr<cv::CLAHE> mClahe = cv::createCLAHE(3.0, cv::Size(4, 4));

private:
    rclcpp::Logger logger_;
    rclcpp::Publisher<geometry_msgs::msg::PoseStamped>::SharedPtr orb_pub_;
    std::unique_ptr<tf2_ros::TransformBroadcaster> tf_broadcaster_;
};
[... 178 lines cut ...]
            pose_msg.pose.orientation.x = quat.x();
            pose_msg.pose.orientation.y = quat.y();
            pose_msg.pose.orientation.z = quat.z();
            pose_msg.pose.orientation.w = quat.w();

            orb_pub_->publish(pose_msg);
        }

        // Control loop frequency
        loop_rate.sleep();
    }
}

void ImuGrabber::GrabImu(const sensor_msgs::msg::Imu::SharedPtr imu_msg)
{
    mBufMutex.lock();
    imuBuf.push(imu_msg);
    mBufMutex.unlock();
}

void ImageGrabber::SetPub(rclcpp::Publisher<geometry_msgs::msg::PoseStamped>::SharedPtr pub)
{
    orb_pub_ = pub;
}
```

astronaut71, 2024-03-06:
hi @Nikolaiarn I posted my repo and some of the nodes I tried. Please let me know if you able to fix it

Nikolaiarn, 2024-03-06:
@astronaut71, thanks for the reply. I will let you know if I get t to work.

TheRealBeef, 2024-05-04:
I believe I have mono-inertial working - it runs but I have no calibration for my IMU to confirm and haven't run any rosbags yet. I will have real-world tests by end of this week anyway when I'm back at the robot

https://github.com/TheRealBeef/ORB_SLAM3_ROS2

I'm currently running with the following command - note I've done "bad" and edited TUM-VI.yaml to match my current webcam. I'll revert that before I call the repo completed.

`ros2 run orbslam3 mono-inertial src/orbslam3_ros2/vocabulary/ORBvoc.txt src/orbslam3_ros2/config/monocular-inertial/TUM-VI.yaml --ros-args --remap /camera:=/camera1/image_raw`

davidxupiaopiao, 2024-05-06:
unsubscribe

 

 
https://github.com/TheRealBeef/ORB_SLAM3_ROS2