UZ-SLAMLab/ORB_SLAM3 #794: Errors with ./build.sh, seemingly with eigen3 and sophus

advait-patel-17, 2023-08-10:
OS: Ubuntu 18.04

Was getting a bunch of errors when running build.sh, so I decided to run each of the commands individually. Building DBoW2 and g2o seem to be fine - no errors. Running `make -j` gives me a bunch of errors that seem to have to do with ceres (pictures attached).

When running `cmake .. -DCMAKE_BUILD_TYPE=Release` in ORB_SLAM3/build I get:

```'Could not find a configuration file for package Eigen3 that is compatible with requested version '3.3.7'. the following configuration files were considered but not accepted:

/usr/lib/cmake/eigen3/Eigen3Config.cmake, version: 3.3.4```

I have downloaded and extracted the tar.gz file from the eigen3 wiki, it is at ~/eigen-3.4.0. I'm not sure how to get it to recognize this version, or if I was supposed to do something else to properly install eigen3.

hanseongbugi, 2023-08-25:
Install eigen3 using the following command.
``
sudo apt update && sudo apt upgrade
sudo apt install libeigen3-dev
``
If you install eigen3 in the same way as above, version 3.3.4 is installed for ubuntu 18.04
You can use the 3.3.4 version to successfully install ORBSLAM3.

GabrielGosden, 2023-09-15:
@

> Install eigen3 using the following command. `sudo apt update && sudo apt upgrade` `sudo apt install libeigen3-dev` If you install eigen3 in the same way as above, version 3.3.4 is installed for ubuntu 18.04 You can use the 3.3.4 version to successfully install ORBSLAM3.

Thank you. My implementation stopped working and uninstalling libeigen3-dev and installing it again fixed the issue. I do not know what I did to break it.