UZ-SLAMLab/ORB_SLAM3 #797: error with ./build_ros.sh, already did a lot of solutions but error persisted

daelisa, 2023-08-21:
Hello! I`m a begginer, so I might be commiting a foolish mistake with my ROS or my instalation in general. I already searched through the Issues and a lot of people seemed to have the same problem as me, and I tried pretty much everything suggested, yet nothing work (although the error message changed, it still exists).

My bashrc currently looks like this 

> export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~/ORB_SLAM3/Examples_old/ROS
> source ~/catkin_ws/devel/setup.bash

I noticed that I don`t have the following path
> ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/build/devel/setup.bash
But I don`t know how to resolve it. Honeslty, I have no idea what I`m doing wrong. I followed the step by step, I`m just lost now, haha.

The error I get when I do the ./build_ros.sh is this

Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:2 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:7 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:16 (cmake_policy):
  The OLD behavior for policy CMP0002 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:18 (cmake_policy):
  The OLD behavior for policy CMP0003 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:20 (cmake_policy):
  The OLD behavior for policy CMP0005 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:23 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Warning (dev) at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:18 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:38 (include)
  CMakeLists.txt:2 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:12 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:40 (include)
  CMakeLists.txt:2 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

[rosbuild] Building package ORB_SLAM3
[rosbuild] Error from directory check: /opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/oem/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3
1
Traceback (most recent call last):
  File "/opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:99 (message):
  [rosbuild] rospack found package "ORB_SLAM3" at "", but the current
  directory is "/home/oem/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3".  You should
  double-check your ROS_PACKAGE_PATH to ensure that packages are found in the
  correct precedence order.
Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:5 (rosbuild_init)

-- Configuring incomplete, errors occurred!
make: *** No targets specified and no makefile found.  Stop.

If someone is able to help me, I would be very grateful! I`m using ROS NOETIC 22.04

aPR0T0, 2023-08-23:
Please clone the fork, everything is solved in that, lemme know if you still face the issue
from the #798

shu1ong, 2024-05-13:
> Please clone the fork, everything is solved in that, lemme know if you still face the issue from the #798

it seems the same problem in the fork