UZ-SLAMLab/ORB_SLAM3 #805: Problem with the visualization of Map Points when using a presaved map

dionkrith, 2023-09-06:
Hello. I am trying to load a pre-saved atlas using LoadAtlas() function. But when I load the binary file the map viewer only shows the keyframes of the camera trajecotry and not the 3D map points. Is there someone that have faced the same problem? Thank you in advance.

Aishkrish18, 2023-09-12:
I am facing the same problem as well. I am not sure how to print the map points. Did you manage to find it?

dionkrith, 2023-09-15:
I managed to find a solution for visualization. I think that in
MapDrawer.cc in the DrawMapPoints() function only the map points of the
active map are going to be visualized. But when you load a pre-saved map
from what I understood orbslam3 creates also a new map and sets this map as
the active map.  The pre-saved map turns out that is saved in the atlas as
a non-active map. That is why we cannot see the map points visualized. You
can change the code a little bit to show not only the map points from the
active map but also the map points from all the maps that are saved in the
atlas.

JiangongChen, 2023-10-30:
Replace 
`mpAtlas->CreateNewMap();`
by
`mpAtlas->ChangeMap(mpAtlas->GetAllMaps()[0]);`
in the code to load map in the declare function in the System.cc will set the loaded map as active map instead of creating a new map.

opentld, 2024-01-23:
> Replace `mpAtlas->CreateNewMap();` by `mpAtlas->ChangeMap(mpAtlas->GetAllMaps()[0]);` in the code to load map in the declare function in the System.cc will set the loaded map as active map instead of creating a new map.

你好，我读取了之前保存的地图，尝试进行定位，但是一直无法定位成功，最终又新建了一个地图，请问，这个问题怎么解决，才能实现读取之前建立的地图，并进行定位呢？

jonaug88, 2024-03-23:
> Replace `mpAtlas->CreateNewMap();` by `mpAtlas->ChangeMap(mpAtlas->GetAllMaps()[0]);` in the code to load map in the declare function in the System.cc will set the loaded map as active map instead of creating a new map.

This works for me ! However, there is one issue i am facing when doing this. It seems to load the map and the keyframes/trajectory just fine, however quite rapidly it seems to lose track of the local map, thus resetting the point cloud and starting over, while keeping the keyframes and the trajectory. 
Did you experience this and/or do you have any ideas how to sort this issue out? 

Thanks in advance