UZ-SLAMLab/ORB_SLAM3 #810: Segmentation fault in Sophus funtion

astronaut71, 2023-09-15:
Hi Im using ORB_SLAM3 in Monocular Inertial mode. When using only Camera ( Monocular ) it works fine, But when use with IMU I got 
Segmentation fault  error. This is the error

```
Sophus ensure failed in function 'Sophus::SO3<Scalar_, Options>::SO3(const Transformation&) [with Scalar_ = float; int Options = 0; Sophus::SO3<Scalar_, Options>::Transformation = Eigen::Matrix<float, 3, 3>]', file '/ORB_SLAM3/Thirdparty/Sophus/sophus/so3.hpp', line 470.
R is not orthogonal:
 1.28446e-07        -nan 3.37115e+19
       -nan        -nan        -nan
3.37115e+19        -nan         inf
```

at the moment I dont have online values for the accelerometer so just put some constant small numbers such as  cv::Point3f acc(0.03, 0.01, 0.001);Does this cost the error or? Should I use also `ros2 run tf2_ros static_transform_publisher 0.01 0.02 0 0.005 0.01 0.01 /world  /camera ? 

Thanks. Any help?
`

SM20598, 2025-02-21:
I have similar issues using https://github.com/zang09/ORB_SLAM3_ROS2/issues?q=orthogonal

SM20598, 2025-02-21:
I think this issue is related to #598.

In my case, I did not encounter nan values, but I get
```
Left camera to Imu Transform (Tbc): 
[0.0070000002, -1, 0, 0.023;
 0.001, 0, -1, 0;
 1, 0.0070000002, 0.001, -0.0020000001;
 0, 0, 0, 1]
Sophus ensure failed in function 'Sophus::SO3<Scalar_, Options>::SO3(const Transformation&) [with Scalar_ = float; int Options = 0; Sophus::SO3<Scalar_, Options>::Transformation = Eigen::Matrix<float, 3, 3>]', file '/colcon_ws/ORB_SLAM3/Thirdparty/Sophus/sophus/so3.hpp', line 470.
R is not orthogonal:
 1.00005   7e-06       0
  7e-06       1       0
      0       0 1.00005
```