UZ-SLAMLab/ORB_SLAM3 #811: D435i Camera-Imu TF Problem

oosuiw, 2023-09-18:
**Summary**
_In the attached TF tree, could it be that excessive TFs increase the buffer, increasing latency and being the main cause of 'resetting'?_
[image]

---

hello,

I am using ORB-SLAM3 in a ROS environment (Ubuntu18.04/ROS Melodic).

I am currently trying to adapt 2 cameras to ORB-SLAM3 ROS Ver. and they are RealSense D435i and D435.

The difference between the two cameras is whether they have an IMU or not, and I decided to develop without an IMU and using RGB-D only.

So, I'm going to use the D435 as-is and remove the IMU from the D435i. 

However, it was the D435i that was having issues, and my guess is that it was the IMU.

I did some searching and found that this symptom is referred to as "resetting". 

When running ORB-SLAM3, I have a symptom where the current map, as represented in the PANGOLIN GUI, is reset at regular intervals of time.

Before resolving this, I tried running the sample data which works fine and everything matches with the TF tree, but when I run the D435i, I get the 'resetting' symptom.

To make this easier to understand, I'm attaching a picture of the camera_pose captured in plotjuggler.

[image]

---
***Here's what you can try***

1. camera-IMU TF
2. calibrate IMU
3. disable IMU node (w/RealSense2 manager_node)
4. change node (ros_rgbd_inertial->ros_rgbd)

The above list is all. 

Other issues say that the IMU is not calibrated, the TimeStamp is not correct, etc... but in my situation, initial_reset = true; and for global_time_enable, I couldn't find the argument (searching within the package didn't yield any results).

So, I'd like to know how to disable the IMU or work around the 'resetting' issue, as I'd like to use it in the same way as the D435 with no difference, rather than calibrate the IMU and use Inertial mode (=IMU) through tasks.

Grange007, 2025-03-24:
Have you figured it out? I have the same problems.

oosuiw, 2025-03-25:
@Grange007 It's been a while, so I don't remember exactly...

As I recall, with the D435i, I noticed a buffer buildup, so I disabled the IMU.

It's not a very good solution and I can't recommend it.

closed 2025-03-25 by oosuiw