UZ-SLAMLab/ORB_SLAM3 #812: Monocular ORBSLAM3 with D345 in REAL TIME

EduardoQD, 2023-09-19:
Hello,
I would like to launch the algorithm but in real time, not reading one folder with frames.

Does anyone know the correct modications of the script "mono_euro.cc" ?

Regards

oosuiw, 2023-09-22:
hello,

I'm replying because I thought it might be helpful if I could confirm that the camera you are using is the D435 and not the D345.

The D435 is an RGB-D camera, so you should use rgbd_realsense_D435i, not mono_euro.cc.

Best Regards

EduardoQD, 2023-09-22:
I'm using  D435 without IMU module.

If I'm not wrong the problem is the same because rgbd_realsense doesn't use frames in real time.

Right ?

oosuiw, 2023-09-22:
I'm actually using the D435 and D435i on ROS. Of course, I'm doing real-time processing.

Before using ORB-SLAM3, I was successful with ORB-SLAM2 to run SLAM mode in real-time on a non-ROS version with a D435.

If you are using ROS, the issue with Frame can be resolved through TF.

Additionally, as far as I know, using Examples/RGB-D/RealSense_D435i.yaml should solve your problem.  

Finally, if you are also considering running on ORB-SLAM2, I would be grateful for any additional responses.

EduardoQD, 2023-09-22:
I'm  not using ROS,
I don't have problem using Orbslam2 or 3 because I would like to use the algorithm in real time.

oosuiw, 2023-09-22:
So I'll ask the question from a different perspective.

Did you run build.sh as described in READ.md to complete the build?

Have you tested the sample using a dataset like EuRoC?

If you do these things in that order, you should be able to get your bearings!

Good luck!

EduardoQD, 2023-09-22:
Yes the algorithm works,

Furthemore I've captured frames and timestamps from the d435 and I've stored theme in frames and I've launched the algorithm.

Now I would like do the same but in real time

oosuiw, 2023-09-27:
I apologize for the delay in responding.

When you say the algorithm works, do you mean you ran the sample data?

Additionally, I would like to ask, what code did you run ORB-SLAM with in the terminal?