UZ-SLAMLab/ORB_SLAM3 #818: How to save the maps?

jumanascimento, 2023-10-10:
is possible to svae he map and the point cloud to use them in later process?

EMP-T, 2023-10-27:
In case you haven't found it already:

`System.SaveAtlasToFile: "./example"`

in the yaml file should be what you were looking for.
It's mentioned on page 9 of [Calibration_Tutorial.pdf](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Calibration_Tutorial.pdf)

shneka-swamy, 2023-10-27:
@EMP-T  Thanks a lot for your reply. But adding the command to TUM_VI.yaml at the last line gives me a Segmentation Fault. Do you have any idea as to what I might be doing wrong?

Thanks in advance!!

jumanascimento, 2023-10-30:
> System.SaveAtlasToFile:

Just something that i did not understood, i should pass this parameter in the yaml file that i am using, but what should a pass to the parameter? Should be a file or the path to de location were i want the map to be saved? it was not clear in the calibration_tutorial. And other thing, this parameter will save the point cloud as well or just the map?

EMP-T, 2023-10-31:
> @EMP-T Thanks a lot for your reply. But adding the command to TUM_VI.yaml at the last line gives me a Segmentation Fault. Do you have any idea as to what I might be doing wrong?
> 
> Thanks in advance!!

Could you provide the line that you added to the yaml file?

EMP-T, 2023-10-31:
> Just something that i did not understood, i should pass this parameter in the yaml file that i am using, but what should a pass to the parameter? Should be a file or the path to de location were i [...]

You have to provide the path to the file that you want the map to be written to. The extension will be added.

jumanascimento, 2023-10-31:
> You have to provide the path to the file that you want the map to be written to. The extension will be added.

Thanks, I managed to make it work. One last thing, he saved the file as a .osa, is that the only save extension or is it possible to save the file as an .obj for example?

unicornuniform, 2023-11-02:
> @EMP-T Thanks a lot for your reply. But adding the command to TUM_VI.yaml at the last line gives me a Segmentation Fault. Do you have any idea as to what I might be doing wrong?
> 
> Thanks in advance!!

Does anyone have a solution to the segmentation fault issue? Inconsistently, it segfaults for me when saving the map

shneka-swamy, 2023-11-02:
The segmentation fault while saving maps (for me) was happening due to the deletion of the items used for iterating through set in `Map::PreSave` and `MapPoint::PreSave`. I changed these functions to iterate and delete on different sets.

I made the following changes as mentioned by @heiwang1997 in https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/443#issuecomment-1154814254 to save the maps.

Hope that helps !!

unicornuniform, 2023-11-04:
> The segmentation fault while saving maps (for me) was happening due to the deletion of the items used for iterating through set in `Map::PreSave` and `MapPoint::PreSave`. I changed these functions to iterate and delete on different sets.
> 
> I made the following changes as mentioned by @heiwang1997 in [#443 (comment)](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/443#issuecomment-1154814254) to save the maps.
> 
> Hope that helps !!

Thanks. Yes from limited testing on few small maps that I built and saved, the fix seems to work, though I haven't tried on a large one. Iterating through the list while at the same time changing its size seems to be the culprit. Hopefully these changes will make it into the official branch but I don't know if it's being maintained anymore

altaykacan, 2024-01-02:
Hey all,

mentioned this on other issues, I shared some code I used to save the point cloud file [here](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/390#issuecomment-1874447943). 

Just sharing it here so you also get notified, I hope it helps!