UZ-SLAMLab/ORB_SLAM3 #819: librealsense2 version error -- CANNOT RUN DEMOS

EricZhang1412, 2023-10-15:
I use D455 camera, and it happened when I ran the demo :

$ ./Examples/Stereo-Inertial/stereo_inertial_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/Stereo-Inertial/RealSense_D435i.yaml

terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  API version mismatch: librealsense.so was compiled with API version 2.50.0 but the application was compiled with 2.54.1! Make sure correct version of the library is installed (make install)
Aborted (core dumped)

HOW CAN I FIX THIS

I do not want to change my librealsense version. Thx!

IAMHAADICOOL, 2023-11-08:
Can you please provide a solution?

EricZhang1412, 2023-12-06:
i lowered the API version to 2.50.0, although it is a bit complicated when configuring the environments. There are many ways to delete the previous version and install the current version one。 Hope this helps

closed 2023-12-06 by EricZhang1412