UZ-SLAMLab/ORB_SLAM3 #821: Can't run ORB_SLAM3

IAMHAADICOOL, 2023-10-18:
The library built successfully with the help of ./build.sh command
Now when I trying to run it using the following command
./Examples/Stereo-Inertial/stereo_inertial_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/Stereo-Inertial/RealSense_D435i.yaml
I am getting the following error./Examples/Stereo-Inertial/stereo_inertial_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/Stereo-Inertial/RealSense_D435i.yaml
[image]
I have connected the Realsense camera successfully, I am not able to understand why is it happening.
Can someone help me out please?

CurlyWei, 2023-10-23:
Hi bro,you execute the .cc file(show terminal pic).
You should execute the exe file

IAMHAADICOOL, 2023-10-23:
I can't find the exe file after building

CurlyWei, 2023-10-23:
1.Can you show or check your document(ORB_SLAM3/Example/Stereo-Inertial/stereo_inertial_realsense_D435i) ?
2.make sure your build.sh is successful ?

IAMHAADICOOL, 2023-10-23:
Yeah, I checked it😅
And yeah, I built it three times, it build successfully
There is just cc file in the mentioned location of yours😅

CurlyWei, 2023-10-23:
Did you have install realsense?
Check  line 213 in the CMakeLists.txt,If you didn`t have it,
the exe can`t be built it

IAMHAADICOOL, 2023-10-23:
Oh yeah, maybe that is the case, I'll built Realsense and then respond back
to you, thanks, I'll surely respond back.

closed 2023-11-10 by IAMHAADICOOL