UZ-SLAMLab/ORB_SLAM3 #823: output stream error

YamadaNaoto712, 2023-10-31:
Hi, when I click "stop button" in order to save map, this error occurs

`Shutdown
Starting to write the save binary file
terminate called after throwing an instance of 'boost::archive::archive_exception'
  what():  output stream error
Aborted`

How can I solve this problem? My yaml file is below.

%YAML:1.0

#--------------------------------------------------------------------------------------------
 Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
File.version: "1.0"

Camera.type: "PinHole"

 Left Camera calibraition and distortion parameters (OpenCV)
Camera1.fx: 569.60760498
Camera1.fy: 569.60760498
Camera1.cx: 630.93981934
Camera1.cy: 366.25228882

 Kannala-Brandt distortion parameters
Camera1.k1: 0.0
Camera1.k2: 0.0
Camera1.k3: 0.0
Camera1.k4: 0.0
Camera1.p1: 0.0
Camera1.p2: 0.0
 Right Camera calibration and distortion parameters (OpenCV)
Camera2.fx: 564.47619629
Camera2.fy: 564.47619629
Camera2.cx: 637.48828125
Camera2.cy: 365.96850586

 Kannala-Brandt distortion parameters
Camera2.k1: 0.0
Camera2.k2: 0.0
Camera2.k3: 0.0
Camera2.k4: 0.0
Camera2.p1: 0.0
Camera2.p2: 0.0
 Transformation matrix from right camera to left camera
Stereo.T_c1_c2: !!opencv-matrix
  rows: 4
  cols: 4
  dt: f
  data: [ 0.9999977087482047, 0.0003130080129498591,  -0.002117669550414781, -0.073462406329529,
        -0.00031342438557307073, 0.9999999316178285, -0.00019628930921337113, 0.0003497394134312251,
         0.002117607965477298, 0.0001969525887428245,  0.999997738470534,  0.0007627635314297477,
         0.0,  0.0,  0.0,  1.0]

 Overlapping area between images (to be updated)
Camera1.overlappingBegin: 0
Camera1.overlappingEnd: 848

         #Camera2.overlappingBegin: 0
         #Camera2.overlappingEnd: 848

 Camera resolution
Camera.width: 1280
Camera.height: 720

 Camera frames per second
Camera.fps: 30
Camera.bf: 42.7205703735
 Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

 Close/Far threshold. Baseline times.
Stereo.ThDepth: 37.5

#--------------------------------------------------------------------------------------------
 IMU Parameters
#--------------------------------------------------------------------------------------------

 Transformation from body-frame (imu) to left camera
IMU.T_b_c1: !!opencv-matrix
   rows: 4
   cols: 4
   dt: f
   data: [0, 1, 0, 0.06942698,
        -1.0, 0, 0, 0.00819243,
        0, 0, 1,  -0.00434384,
        0.,        0.,        0.,        1.      ]

 Do not insert KFs when recently lost
IMU.InsertKFsWhenLost: 0

 IMU noise
IMU.NoiseGyro: 1.6459319892541075e-03 # 0.000005148030141 # rad/s^0.5
IMU.NoiseAcc: 1.9294825259447989e-02 # 0.000066952452471 # m/s^1.5
IMU.GyroWalk: 5.6469374967339070e-05 # rad/s^1.5
IMU.AccWalk: 3.7415914816775128e-04 # m/s^2.5
IMU.Frequency: 400.0

#--------------------------------------------------------------------------------------------
 ORB Parameters
#--------------------------------------------------------------------------------------------

 ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1200 # Tested with 1250

 ORB Extractor: Scale factor between levels in the scale pyramid
ORBextractor.scaleFactor: 1.2

 ORB Extractor: Number of levels in the scale pyramid
ORBextractor.nLevels: 8

 ORB Extractor: Fast threshold
 Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
 Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
 You can lower these values if your images have low contrast
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
 Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1.0
Viewer.GraphLineWidth: 0.9
Viewer.PointSize: 2.0
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3.0
Viewer.ViewpointX: 0.0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500.0
Viewer.imageViewScale: 1.0

 Atlas parameters
#--------------------------------------------------------------------------------------------
System.SaveAtlasToFile: "/home/yamada/premap.osa"
#--------------------------------------------------------------------------------------------

Any information makes me happy! Thank you.

closed 2023-10-31 by YamadaNaoto712

mrhen2324, 2023-11-15:
I get the same error message "Shutdown Starting to write the save binary file terminate called after throwing an instance of 'boost::archive::archive_exception' what():  output stream error Aborted". Could you solve the Problem?

mrhen2324, 2023-11-15:
Thanks, that works. I have received a premap.osa file. Which programme do you use to open it? Do you get a 3d view of the point cloud?

YamadaNaoto712, 2023-11-15:
Unfortunately, I have not found how to open .osa file. So I change the code of MapDrawer.cc included in `ORB_SLAM3/src`
please refer to [https://blog.csdn.net/qq_45848817/article/details/126024785](url)