UZ-SLAMLab/ORB_SLAM3 #826: Why can't i achieve the accuracy in the paper

gxf193558, 2023-11-06:
When I used the euroc data set to run monocular and stereocular, I used the evo tool to evaluate and found that the rmse was basically greater than the value in the paper, and the error in the paper was in the decimeter level. For example, for the mh01 sequence, the rmse result I got in the mono case was 0.043 while that in the paper was 0.016

shneka-swamy, 2023-12-04:
Hey,

I am facing the same issue with TUM-VI dataset as well. Did you get any feasible solutions?

Mechazo11, 2023-12-11:
@gxf193558 and @shneka-swamy if I recall correctly, the authors mentioned using a "processed" ground-truth. You may want to procure that. In my experiments, using the evo toolbox, the best rmse I got was 0.051 m averaged over 10 runs.  I did not use the "processed" ground truth and directly used the ones I got from EuRoC MAV without any modifications to them.